EverythingSmartHome/everything-presence-lite #140: Odd behaviour with illuminance

gazzaman22, 2024-08-17:
Has the sun goes down the reading from the illuminance sensor progressively goes down  until it gets around a reading of 65 then starts going back up (300 )and behave erratically until i get a reading from the sensor of about 30 
[image]

clintebbesen, 2024-08-19:
is it by chance the sun reflecting off something? maybe as the sun gets lower in the sky it comes directly through a window and shines on the wall behind the sensor or directly on the sensor or something like that? whereas when the sun is higher in the sky the wall behind the sensor or the sensor itself might be in shade.

gazzaman22, 2024-08-19:
I don't believe so as the sun is behind other houses at this time but I will go and move the sensor to another room  at the other  side of the house to check it out

markwebsterUK, 2024-08-19:
This is since firmware 1.0.6, I'm seeing the same. Luminance based automations have become unusable for me. I've downgraded to 1.0.5 and it's all rosy again.

gazzaman22, 2024-08-19:
Thanks mark I will give that ago

GurbEHV, 2024-08-26:
I have issues with illuminance being all over the place as well. Here's the history for this morning:
[image]

The "slaapkamer" and "werkkamer" are definitely not hit by direct sunlight; the "woonkamer" one possibly but also unlikely in the morning. The "slaapkamer" is an eastward facing room, "werkkamer" a westward facing room, "woonkamer" has windows on both east and west.

markwebsterUK, 2024-09-04:
That's exactly what it is, I came to the same conclusion here https://github.com/EverythingSmartHome/everything-presence-lite/issues/137

To revert to an older version, edit the yaml in ESPHome and change the packages section to:
```
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@1.0.5
```
It's been working flawlessly for me since I did this, so I won't be updating for now.

iay, 2024-09-04:
Thanks, I will do some measurements (I've found a dimmable light that looks like it would work to test, although it turns out the batteries for my old Gossen light meter were discontinued 20 years ago...) and then retry with the old version.

This definitely looks like a bug to me, though; I concur with your understanding that both the left and right series of numbers should be monotonic.

I don't care at all about absolute accuracy, although I understand some people might, so another thing I might try is just forking the latest code and removing that filter entirely (always assuming that one can just point that reference to anywhere on GitHub without anything getting confused?)

clintebbesen, 2024-09-04:
@iay rather than removing that code, it would be quicker, easier and more reliable to just adjust the values in the right column to be linear. 

Also rather than forking, maybe if you made a branch here, created a pull request @EverythingSmartHome might be able to reasonable quickly accept that one code change and push that live for everyone as linear values would seem to be better than what is currently there and then in future they can revisit what values they actually want

iay, 2024-09-05:
> @iay rather than removing that code, it would be quicker, easier and more reliable to just adjust the values in the right column to be linear.

It doesn't look that way to me, honestly.

> Also rather than forking, maybe if you made a branch here, created a pull request @EverythingSmartHome might be able to reasonable quickly accept that one code change and push that live for everyone [...]

I can't create a branch here, it's not my repository. If I was going to create a PR, I would need to do it from my own repository, created by forking. This is not a big deal, there are [41 forks](https://github.com/EverythingSmartHome/everything-presence-lite/forks) of this repository already.

EverythingSmartHome, 2024-09-05:
Just background on the issue - it is indeed likely to do with the calibration values. The calibration values may look like there was a typo, however the values are correct. The reason there can be big difference between one set and similar readings between another set is because two units were tested side by side with each other, one with the cover on and one with it off.

I then did a bunch of testing with the units in different rooms, different light sources, different positioning of the light source etc. When the light source is in front of the unit, there is of course a big difference with the cover on vs off. When the light source is above, there is much less of a difference because of the top vent holes. So that's the reason some will be close and others not.

I'm not sure if this is the right approach and it just needs refining, it made sense to me to test in that manner since users will all have different positioning.

Feel free to submit a PR! One idea is to change the method from exact to least_squares which would smooth it out. Would love if someone is able to test if they have a re-producible environment, unfortunately it's not easy for me to reproduce this currently.

iay, 2024-09-05:
I did the experiment I was talking about; here are the results for anyone who is interested:

[image]

This was all done by placing the unit about 18" from a dimmable quartz halogen floor lamp I have, the idea being that the glowing incandescent stuff doesn't flicker as much as an LED might so there's less chance of confusing the sensor. I started at dark, and every 10 seconds tweaked the dimmer knob as little as possible towards the brighter end. The whole ramp takes about 10 minutes.

The first ramp is with 1.0.9 and you can see there's a point where the output value jumps to about 350, corresponding to an input value around 14.

I thought I did the second ramp after re-flashing to 1.0.5, but it turned out that I had just downloaded the same firmware again because I hadn't done a "Clear Build Files". This time I caught the round-14 excursion as well as something a bit higher up which is probably one of the flanks of the 21/885 mountain.

The third ramp was with 1.0.5 and you can see it's perfectly smooth. It tops out at around 370 instead of 480, but that's what you'd expect from that part of the calibration table (between  283 -> 394 and 807 -> 993).

EverythingSmartHome, 2024-09-05:
@iay Good test, thanks! Since you have a reproducible test, is it possible you can try changing the method to least squares an seeing how that performs?

markwebsterUK, 2024-09-05:
> I then did a bunch of testing with the units in different rooms, different light sources, different positioning of the light source etc.

You can't account for all of that with a single set of values though, it'll just end up being wrong for everyone. I don't think you even should be trying to calibrate at all. People will put these things in place and adjust any lux based automations depending on the light their unit gets in that position. The value jumping around like it does now makes it impossible to do that. A predictable `more light = higher value` is much more important that trying to get perfectly correct values.

iay, 2024-09-05:
> @iay Good test, thanks! Since you have a reproducible test, is it possible you can try changing the method to least squares and seeing how that performs?

My test is only reproducible when it's dark outside, so it will have to wait until tonight :)

I'm also going to have to figure out how to actually make code changes in a fork, but I imagine that's fairly simple.

However, I am pretty sure that changing the method to least squares WILL change the output but WILL NOT fix the underlying issue. You will still have output which doesn't increase monotonically with increasing input light.

Calibration is about making sure the output is accurate under some specific test conditions. You can't simultaneously calibrate a device for multiple conditions at once, it's just not physically possible. The correct answer for a given raw sensor value will be different in different conditions and a single table cannot - by definition - encode multiple possibilities.

I think you have three alternatives:

1. Take the calibration filter out and tell people that it's uncalibrated. I guess you had that originally and people complained?
2. Set up a single specific test setup: so-and-so light temperature, so-and-so distance, cover on, light from the front, whatever. Take readings and make a calibration table. Tell people what specific situation the device is calibrated for.
3. Set up multiple test setups, build multiple calibration tables and provide a control to allow the deployer to select the filter most appropriate to their use case. Rather you than me.
3a. Provide a control to disable the calibration and just give raw values.

As @markwebsterUK is saying, for almost all uses the "more light means bigger number" guarantee is going to be more important than absolute accuracy. Any calibration table which jumps up and down is going to fail that test. Whether an individual customer notices this will depend very much on their situation (if it's just "curtains are open" then possibly not).

EverythingSmartHome, 2024-09-05:
@iay Oh, I'm absolutely in agreement that more light should produce a bigger number, 100%. The calibration data is specifically to account for the front cover being on. I personally had no issues with raw values because it was super predictable which is what you want with light data. However there was a small number of users who seemed to have it in a certain position in a room where the difference between night and day was small. I'm guessing it was in a position where something was above and below and blocking a lot of the light.

I think the least_squares method should average out the calibration values, but be interesting to test. I'll try and see if I can get a re-producible way to test it here.

Otherwise I think starting to remove some of the calibration values may be best place to start to see if we can get a more smooth and proper set of values.

markwebsterUK, 2024-09-05:
> What makes you say that?
Because if you have an indirect light source, you're still going to be subject to the calibration values for the direct light source, and vise versa.

> some users have it in such a position in a room where the difference between night and day is 10-15 lux, making it difficult to create automations with such a low differential.

I'm not sure what the capabilities of the sensor are, but can it report more than one decimal place? That should help in situations where there is a small difference between night and day. Otherwise some kind of optional multiplier could work.

iay, 2024-09-06:
Last night I was able to flash my unit with a copy of 1.0.9 with the `calibrate_linear: method:` set to `least_squares`. As I expected, this doesn't give the right answer but I will admit that it failed in a way I didn't expect. The flip side is that I think I understand what they are doing a bit better.

Here's the run:

[image]

The ramp is now monotonic as the incoming light increases (this is good) but the baseline has lifted to around 165lx (this is bad).

What I think is happening is that:

- `method: least_squares` is trying to make a map from input to output with a single linear (`y=mx+c`) segment which minimises the error vs. the collection of input points provided. It's not guaranteed to pass through any of the provided points. This is the sort of thing you'd want if you know the curve is supposed to be linear but your data points are scattered. In this case, where the data isn't just noisy but covers multiple scenarios, the effect is that the bad outliers in the data pull the baseline (`c`) up and probably mess up the slope (`m`) as well.
- `method: exact` was, I think, doing something else: making a map from multiple linear segments connecting each of the provided points. This is what you'd want if you think the curve is NOT linear, but can be approximated by a number of linear segments AND you have accurate (not noisy) data for various points across the range. Again, this is not what we have: this algorithm slavishly puts the curve through every point so the outliers cause weird excursions.

Looking at the datasheet, it seems that there are two things the manufacturers are hot on:

1. In the dark, the sensor will say 0lx
2. The output is very linear vs. incoming light

This suggests that we can get a more accurate answer with a simpler setup: calibration can be linear (`y=mx+c`) but we know that `c` is 0 so we only need one measurement to find `m`. We can apply the multiplier either in a `multiply` filter or in the `lambda` where the offset is applied. The latter would have the advantage that we could have an adjustable `m` scaling factor as well to allow people to calibrate for their own setup: the data sheet implies that there's a fair amount of variation between devices (about 45%).

As to a factory calibration in the firmware, it seems to me that the easiest thing to achieve would be to set `m` to be a multiplier which compensates for the case. I think that would be as simple as (a) disabling the calibration entirely (b) taking the top off (c) shine a light on the unit and record the resulting value A (d) put the top back on and record the resulting value B (which will be smaller). `m` is then `B/A` and once it is applied the result should mimic what the bare sensor sees.

All "I think" obviously.

iay, 2024-09-06:
That reasoning probably depends on lux being a physical unit, but now I come to think of it, it may actually be something that's weighted perceptually. So maybe that wouldn't work very well and we'd be back to the `exact` method with multiple data points as being the only way to do this accurately. Might be worth a try nevertheless.

4D4M-Github, 2024-09-08:
@EverythingSmartHome 
> I then did a bunch of testing with the units in different rooms, different light sources, different positioning of the light source etc. When the light source is in front of the unit, there is of [...]

Oh...am I understanding this correctly? Did you merge two completely different sets of measurements into one calibration table?

I am quite sure that this is not the way forward. 

It is possible to introduce a logical switch that allows users to change the scale depending on whether the lid is on or off, but even this cannot force the sensor to give a higher resolution reading.

The physical limitations of the sensor are given:
[image]
[image]
*When using BH1750

The problem in dark rooms is caused by the sensor's modest capabilities in low light. If you zoom in on this region of the characteristic curve, you will see only steps and control uncertainty.

So **I respectfully suggest to restore either the raw data reporting or at least a monotonically varying calibration curve**.

ps.: If you manufacture the EPL with this sensor(BH1750), can the measurement modes(high-res/) be changed afterwards or "hardwired"?
[image]

EverythingSmartHome, 2024-09-08:
> The physical limitations of the sensor are given:
I think you are possibly misunderstanding the reference to calibration here (since your posting the datasheet which I don't think applies to this). It's not calibrating the actual light sensor. It's an ESPHome calibrate linear function which takes the raw values and applies a filter. The raw values coming from the sensor are not touched or calibrated. The ESPHome filter just reads that raw value, then applies the calibrated values and sends that value to Home Assistant.

> ps.: If you manufacture the EPL with this sensor(BH1750), can the measurement modes(h-res/h-res2 etc) be changed afterwards or "hardwired"?

Measurement modes are set in software I believe, and is done by ESPHome. You can see it in the code here: https://github.com/esphome/esphome/blob/18a1191e03578fb395156cb2578dc2f41d2a44f5/esphome/components/bh1750/bh1750.cpp#L27

I believe the default is Hres2

Thanks for the discussion everyone, it's certainly useful. I've found a method to reproduce this now in a light controlled room so hopefully we can figure out a good solution for everyone. I'm still unsure why a similar method is applied to the EP1 and has been for a long time which has no issues, but will figure that out soon I'm sure!

4D4M-Github, 2024-09-08:
I don't think I misunderstood, but that's probably irrelevant now.

The key moment, which is partly why I quoted the datasheet, is that only a strictly monotonic function can be used in this situation compared to reality. Any deviation from this can only be due to measurement error. Whatever function we try to fit to the raw measurement data must be monotonic.

> I believe the default is Hres2

Then, unfortunately, in the situations indicated, users have reached the limits of the sensor and software wizardry cannot be expected to produce much better results. (Splitting the entire _dark near_ range is only a virtual success, there is no real resolution behind it.)

iay, 2024-09-08:
> I believe the default is Hres2

It looks like (every time round) it samples quickly at low res, and then if the result it gets is lower than 7000lx (which I'd expect it to be) then it re-samples with the slowest and most accurate hi-res 2 mode. So the driver is using the highest resolution available below 7000lx, and backs off to lower resolution for very bright situations.

EverythingSmartHome, 2024-09-16:
Pushed a new version 1.0.10 which reverts the calibration values and applies none till I can figure out a better way to do it if possible. I'll close this one as assume reverting back to the raw values will help but please re-open if not.

closed 2024-09-16 by EverythingSmartHome

iay, 2024-09-16:
@EverythingSmartHome Thanks. I imagine this will improve things; I'll try and test this evening to make sure.

I have a branch which does this plus adds a controllable scaling factor which just multiplies the raw value before adding the offset. I didn't bother making a PR as in the limited testing that I've had time to do (removing the lid and comparing a couple of numbers) it didn't seem to make much difference. But if you're interested it's here:

https://github.com/iay/everything-presence-lite/blob/scaled_values/common/everything-presence-lite-base.yaml#L64-L88

sfranky, 2024-09-16:
> ID: <EverythingSmartHome/everything-presence-lite/issues/140/2352471923@
> github.com>

When will that show up on the website ? It s still stuck a 1.0.9 for me.

iay, 2024-09-16:
@EverythingSmartHome do you need to push the 1.0.10 tag/release for it to show up in ESPHome?

EverythingSmartHome, 2024-09-16:
No sorry don't need to tag release (have done now though) but forgot to run a Github action to deploy to GitHub pages. Should be showing now in a few mins