EverythingSmartHome/everything-presence-lite #144: Can't connect to wifi.

jgdf42, 2024-08-20:
My device is seen in HAS out of the box, but when connecting to it (Improv via BLE), it doesn't want to connect to wifi (2.4 or 5ghz) and fails. I'm on 2024.8.0 core and OS 12.4.

Shadyfox911, 2024-08-22:
> My device is seen in HAS out of the box, but when connecting to it (Improv via BLE), it doesn't want to connect to wifi (2.4 or 5ghz) and fails. I'm on 2024.8.0 core and OS 12.4.

I'm having the same issue, I recently added a node to my ASUS AI Mesh network and now it won't connect to the 2.4ghz network.

EverythingSmartHome, 2024-08-22:
Are you able to provide any logs please?

Shadyfox911, 2024-08-22:
> Are you able to provide any logs please?

Log downloaded after failed WIFI connection

I tried downloaded the log txt file but when I open it, it's blank!

Had to C&P

[D][sensor:094]: 'Target 1 X': Sending state 140.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 200.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 244.13112 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -34.99202 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 138.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 194.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 238.07562 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -35.42577 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 135.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 188.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 231.44978 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -35.68156 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 133.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 182.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 225.41739 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -36.15818 ° with 0 decimals of accuracy
[I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[D][sensor:094]: 'Target 1 X': Sending state 131.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 178.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 221.00905 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -36.35140 ° with 0 decimals of accuracy
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.8lx
[D][sensor:094]: 'Illuminance': Sending state 5.05915 lx with 1 decimals of accuracy
[D][sensor:094]: 'Target 1 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 130.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 175.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Speed': Sending state 0.08000 m/s with 2 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 218.00229 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -36.60707 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 127.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 216.22673 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -35.96893 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 126.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 177.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 217.26712 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -35.44571 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 132.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 173.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 217.60744 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -37.34384 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 138.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 170.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 218.96118 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -39.06847 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 142.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 167.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 219.20995 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -40.37449 ° with 0 decimals of accuracy
[D][bh1750.sensor:159]: 'Illuminance': Got illuminance=1.7lx
[D][sensor:094]: 'Illuminance': Sending state 4.74518 lx with 1 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 140.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 176.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 224.89108 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -38.50065 ° with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 X': Sending state 134.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Y': Sending state 193.00000 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Distance': Sending state 234.95744 mm with 0 decimals of accuracy
[D][sensor:094]: 'Target 1 Angle': Sending state -34.77227 ° with 0 decimals of accuracy
[... cut ...]
I (577) heap_init: At 3FFB7C10 len 000283F0 (160 KiB): DRAM
I (583) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (589) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (596) heap_init: At 40094F30 len 0000B0D0 (44 KiB): IRAM
I (603) spi_flash: detected chip: gd
I (606) spi_flash: flash io: dio
W (610) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (625) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

Shadyfox911, 2024-08-22:
It must be something to do with my network as I tried it on another 2.4ghz network and it connects no problem!

Have tried disabling ax mode and still won't connect

Shadyfox911, 2024-08-22:
Ok so just re-enabled 802.11ax mode (and changed authentication method from WPA2 to WPA2/WPA3 - can't see this making a difference) and it's finally connected 👍

jgdf42, 2024-08-22:
Mine also just decided to work today on my GT-AX6000's 2.4ghz network. Knock on wood, seems alright now.

closed 2024-08-22 by jgdf42