EverythingSmartHome/everything-presence-lite #146: EPL stuck on Detected, Sensor values do not change on movement

ThiSaadeh, 2024-08-21:
I've got 3 EPLs and they have been working pretty well for the last 4 months or so, but recently after one of ESPHome updates they were all "stuck" on Occupancy = Detected while ALL the Sensors values/attributes do not change, no matter how much movement is there within the room the EPL is placed in.

I ended up "reinstalling" ESPHome in all of them, one resumed working properly and detecting motion while the other 2 remain "stuck".

[image]

[image]

[image]

[image]

EverythingSmartHome, 2024-08-22:
Can you download the HLKRadarTool on iOS or Android and try updating the mmwave firmware to 2.04 to see if that helps?

ThiSaadeh, 2024-08-23:
> Can you download the HLKRadarTool on iOS or Android and try updating the mmwave firmware to 2.04 to see if that helps?

I downloaded it and as soon as I got home I was going to test it but something else "fixed" the issue, we were out of power for like 30 minutes throughout the day and when everything powered on again it resumed working correctly.

grayfrost90, 2024-09-05:
I've also got this issue, and the numbers are showing pretty much identical:

[image]

I've updated the firmware on the mmwave but this hasn't changed anything. I think the issue started when I updated the device to esphome 2024.8.0. I want to try downgrading to 2024.7.x but as of yet haven't found the instructions to do that

grayfrost90, 2024-09-05:
I also had a power cut.... but that didn't fix my issue.

should I raise a new issue or can we reopen this one @EverythingSmartHome ?

grayfrost90, 2024-09-05:
actually, I thought I'd try manually pulling the power from the device and plugging it back in.... that does seem to have sorted it

EverythingSmartHome, 2024-09-05:
What version is the mmWave sensor on?

grayfrost90, 2024-09-05:
i _think_ i upgraded it to 2.04 - but it's the messages in the app were all in Chinese so I'm not sure :D

ThiSaadeh, 2024-09-08:
Mine stopped working again, using HLKRadarTool I updated the sensors to 2.0.4 and the EPL resumed working, will test for a week or so and reply back.

I reopened the issue as we still seem to be chasing a solution.

EverythingSmartHome, 2024-09-08:
> i _think_ i upgraded it to 2.04 - but the messages in the app were all in Chinese so I'm not sure :D

What does the "mmWave firmware" sensor in Home Assistant report? The symptoms with those values and stopping responding are all symptoms of the old firmware

grayfrost90, 2024-09-08:
> > i _think_ i upgraded it to 2.04 - but the messages in the app were all in Chinese so I'm not sure :D
> 
> What does the "mmWave firmware" sensor in Home Assistant report? The symptoms with those values and stopping responding are all symptoms of the old firmware

It just says 'unknown'

ThiSaadeh, 2024-09-09:
> > i _think_ i upgraded it to 2.04 - but the messages in the app were all in Chinese so I'm not sure :D
> 
> What does the "mmWave firmware" sensor in Home Assistant report? The symptoms with those values and stopping responding are all symptoms of the old firmware

Is this what you are looking for?

text_sensor": [
        {
          "object_id": "mmwave_firmware",
          "key": 1433581076,
          "name": "mmWave Firmware",
          "unique_id": "sensor-de-presenca-cozinhatext_sensormmwave_firmware",
          "disabled_by_default": true,
          "icon": "",
          "entity_category": 0,
          "device_class": ""

kevinschweikert, 2024-09-12:
I have the same issues with my two EPLs. Disconnecting them and reconnecting them after a few minutes make them work for a while but they get in the stuck state sooner or later

EverythingSmartHome, 2024-09-12:
> > > i _think_ i upgraded it to 2.04 - but the messages in the app were all in Chinese so I'm not sure :D
> > 
> > 
> > What does the "mmWave firmware" sensor in Home Assistant report? The symptoms with those values and stopping responding are all symptoms of the old firmware
> 
> It just says 'unknown'

If it says unknown then most likely not, should say 2.04

EverythingSmartHome, 2024-09-12:
> I have the same issues with my two EPLs. Disconnecting them and reconnecting them after a few minutes make them work for a while but they get in the stuck state sooner or later

Did you confirm update?

ThiSaadeh, 2024-09-12:
Hey Lewis, I have 3 EPLs, two updated successfully and one wont even show up on HLKRadarTool, is there a way to reset it so it shows up?

For those wondering, right in the radar list in the app it is supposed to show the version, here's a snapshot from mine (that worked).

[image]

kevinschweikert, 2024-09-12:
> Did you confirm update?

I tried but the app crashes all the time when I select the found device...

EverythingSmartHome, 2024-09-12:
> Hey Lewis, I have 3 EPLs, two updated successfully and one wont even show up on HLKRadarTool, is there a way to reset it so it shows up?
> 
> For those wondering, right in the radar list in the app it is supposed to show the version, here's a snapshot from mine (that worked).
> 

There is a toggle switch in home assistant to enable Bluetooth on the mmWave sensor itself, guessing yours is off? Its a hidden entity by default so try unhiding it, switching it on and should be good to go

EverythingSmartHome, 2024-09-12:
> > Did you confirm update?
> 
> I tried but the app crashes all the time when I select the found device...

Try the RadarTools app, also made by HiLink, you may have better luck with that. Both can be pretty temperamental!

kevinschweikert, 2024-09-12:
> Try the RadarTools app, also made by HiLink, you may have better luck with that. Both can be pretty temperamental!

That app is so much better! I couldn't see which version they were on but I upgraded them to 2.04. HA mmWave Firmware sensor still reports "unknown" but I still hope that this will fix the issue. Will report back, when I have problems again. Thanks for the quick responses!

[image]

ThiSaadeh, 2024-09-12:
> > > Did you confirm update?
> > 
> > 
> > I tried but the app crashes all the time when I select the found device...
> 
> Try the RadarTools app, also made by HiLink, you may have better luck with that. Both can be pretty temperamental!

Works the same way as the other one, it only shows 2 out of 3 EPLs that I have.

If I swapped the mmWave board to a different EPL would it work?

EverythingSmartHome, 2024-09-13:
> Works the same way as the other one, it only shows 2 out of 3 EPLs that I have.
>
> If I swapped the mmWave board to a different EPL would it work?

Did you try the steps mentioned above to enable Bluetooth on the mmWave?

ThiSaadeh, 2024-09-13:
> Did you try the steps mentioned above to enable Bluetooth on the mmWave?

I tried, but via the Settings >> Devices & Services >> ESPHome >> EPL Device/Entity

Then it displays the following message. Is there any other way I should be going to trigger this to activate?

[image]

Cause even after clicking and getting the error above the Radar tool wont show the missing EPL.

perseus177, 2024-09-15:
I have same problem 
When I go to the room for e.g. 5 seconds - it will clear sensor immediatelly but
When I go to the room for e.g 5 minutes - it will clear sensor after 2-3 minutes

EverythingSmartHome, 2024-09-16:
> P.S.: Shouldn't all of those options be actual buttons upon installation?

No because they aren't generally needed and can confuse new users. For example, even though its called "mmWave Bluetooth", some users think it's to turn on and off Bluetooth proxy. 

Also the Everything Presence Lite Firmware should be left hidden, otherwise you may overwrite the firmware and delete your encryption key. That's the reason it's hidden by default

EverythingSmartHome, 2024-10-08:
I believe this to be resolved but let me know if its still an issue

closed 2024-10-08 by EverythingSmartHome

Kwull, 2024-12-10:
Fixed by adding to config  power_save_mode: none

jouster1974, 2024-12-21:
Ive noticed one of my EPL's has started with the same issue...