EverythingSmartHome/everything-presence-lite #147: Stuck in a boot loop on first reboot after flashing firmware

NicholasDreyer, 2024-08-22:
I have revision 1.2 of the Everything Presence Lite, which worked out of the box flawlessly through Bluetooth pairing; didn't need to flash firmware the first time around. Since purchasing the C4001 (SEN0609) from the Everything Smart Home website, installing it, and flashing the device using the tool on [this page](https://everythingsmarthome.github.io/everything-presence-lite/updating.html), I encounter a boot loop as soon as it reboots. After flashing it, I am able to connect to a Wi-Fi network no problem, Home Assistant recognizes the device and occupancy detection/LED control works perfectly fine. Below is what I see in Home Assistant for visible entities, which appear to be a lot less than the included LD2450 sensor but I am not 100% sure if that is expected.
[image]

Here are the logs from the device with the SEN0609 firmware and sensor installed (I believe this is without bluetooth proxy), which seem to indicate a watchdog timeout based on what I can find through Google but am not very familiar with the ESP32 platform so I am unsure how I could modify the timeout configuration to test that myself.
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/16706226/esp-web-tools-logs.txt)

So far I have tried flashing with a different USB cable that is known to work (I can flash my ratgdo just fine), I tried another computer just in case, different USB ports, flashed with bluetooth proxy and without but they both encounter the issue, swapped the LD2450 back in and flashed the corresponding firmware but that encounters the same behavior just with a lot more in the logs which would be a hassle to censor out right now; when the firmware is flashed back to the LD2450 version, I do see the extra entities come back and they seem to work as expected until reboot, so it recognizes the sensor properly.

If I don't connect to Wi-Fi I don't see the boot loop happen but it's pretty useless to me without being connected to Home Assistant.

I'm at a loss as to what to try next.

EverythingSmartHome, 2024-08-22:
Hello,

 Do you have internet blocked on the device? If so, there is still a bug in esphome which can give a boot loop when internet is blocked. This is due to the http_request component for the new manufacturers update method which is why you won't see it in other esphome devices.

NicholasDreyer, 2024-08-22:
Hi @EverythingSmartHome!

I don't believe so, I'm not even sure how I would do that. When I flash the firmware it connects to the Wi-Fi network just fine, gets recognized within Home Assistant, and can be controlled from Home Assistant up until that first reboot. The screenshot I provided was after a first reboot after disconnect but here is a new one before the first reboot.
[image]

If internet is blocked on the device, it wouldn't even be able to connect for the first time, right? From what I can tell from the logs, it is able to reconnect after reboot and get its IP address from the DHCP server before hitting the watchdog issue.

EverythingSmartHome, 2024-08-22:
Which version of the firmware are your other Lite's currently on? This was due to a feature added in 1.0.5, so if your other Lites are on an older version and haven't been updated, you wouldn't see this issue on those units.

I'm talking about internet access for the devices, not Home Assistants connection to the device. This could be from blocking DNS or firewall rules or network setup.

Are you able to provide logs of the boot loop occuring?

Jello46, 2024-08-22:
I too am seeing the EPL sensor reboot on attempting to update to 1.0.9.

Here is where the reboot happens in the log:
[C][api:025]: Setting up Home Assistant API server...
[D][button:010]: 'get_mmwave_firmware' Pressed.
[D][switch:012]: 'mmWave Configuration Mode' Turning ON.
[I][app:062]: setup() finished successfully!
E (11453) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[16:43:53]E (11453) task_wdt:  - loopTask (CPU 1)
[16:43:53]E (11453) task_wdt: Tasks currently running:
[16:43:53]E (11453) task_wdt: CPU 0: IDLE0
[16:43:53]E (11453) task_wdt: CPU 1: IDLE1
[16:43:53]E (11453) task_wdt: Aborting.
[16:43:53]
[16:43:53]abort() was called at PC 0x401051dc on core 0
[16:43:53]
[16:43:53]
[16:43:53]Backtrace: 0x40082966:0x3ffbe3f0 0x400896e5:0x3ffbe410 0x4008f89a:0x3ffbe430 0x401051dc:0x3ffbe4a0 0x400832e9:0x3ffbe4c0 0x401797bf:0x3ffb2bd0 0x4010592d:0x3ffb2bf0 0x4008aabc:0x3ffb2c10
[16:43:53]
[16:43:53]
[16:43:53]
[16:43:53]
[16:43:53]ELF file SHA256: xxxxxxxxxxx
[16:43:53]
[16:43:53]Rebooting...

Is this enough?

EverythingSmartHome, 2024-08-22:
This looks very similar to the logs when internet is blocked - can you confirm if that is the case?

Jello46, 2024-08-22:
I can confirm the internet was blocked on this device. I unblocked the internet and the device booted fully...

"there is still a bug in esphome which can give a boot loop when internet is blocked. This is due to the http_request component for the new manufacturers update method which is why you won't see it in other esphome devices."

I know you don't control ESPHOME, but blocking iot device is a common practice, any suggestions for a fix?

EverythingSmartHome, 2024-08-22:
Can you confirm how you are blocking or share the firewall rules or method for the block? The reason it's not easy to fix is:

1. It's a niche and new feature
2. I'm not able to re-produce it and either is Jesse - all methods we try of blocking, the device works fine and flags the error properly. We discovered previously that a firewall drop vs reject rule would cause the issue and we believe that to be resolved, so there must be another small nuance that needs resolving, but difficult when we cant reproduce on our end

Jello46, 2024-08-22:
I am blocking via pfsense firewall rule which silently drops the packet.

NicholasDreyer, 2024-08-22:
Ah, my other devices are on 1.0.3 so that explains why they don't encounter the issue.

I temporarily removed all firewall block rules so that wasn't the problem either.

Double-checked my local AdGuard Home DNS server and saw no requests from any of my EPLs; changed the primary DNS in my router, rebooted the EPL, now it works as expected!

Seems like the DNS packets are being dropped somewhere for me. @EverythingSmartHome I can't find documentation indicating what type of DNS the EP1/EPL uses; do you know if it uses plain DNS, DNS-over-HTTPS, DNS-over-TLS, or DNS-over-QUIC?

EverythingSmartHome, 2024-08-27:
> Seems like the DNS packets are being dropped somewhere for me. @EverythingSmartHome I can't find documentation indicating what type of DNS the EP1/EPL uses; do you know if it uses plain DNS, DNS-over-HTTPS, DNS-over-TLS, or DNS-over-QUIC?

That would be a question for ESPHome - I believe it's regular DNS, as the encrypted methods would probably add quite a lot of overhead to a small embedded device but best to check with the ESPHome project as we use the standard wifi component

closed 2024-08-27 by NicholasDreyer