EverythingSmartHome/everything-presence-lite #148: High illuminance readings

V4ler1an, 2024-08-22:
One of my EPLs reads very high illuminance, getting up to 300 when according to my other EPLs (which are consistent) it should be more like 10 to 15. Also, the reading varies quite a lot. When I try adjusting the offset, although I can get it closer to where it should be, the reading wanders around too much to be of any real use. Is there anything I can do? Or is this one defective? Here are the detals, thanks.

[image]

EverythingSmartHome, 2024-08-22:
No, it seems like the devices you have are on different versions. In newer versions of the firmware, there is calibration data added to the light sensor in order to provide better readings when the front cover is on. If you update the other units, you should see similar values - though comparing devices in different rooms is still going to give different readings naturally

V4ler1an, 2024-08-22:
I haven't got round to the firmware updates on the other yet and looking back, the higher readings started after updating the kitchen. I'll get them done. Thank you.

EverythingSmartHome, 2024-08-22:
You are welcome!

closed 2024-08-22 by EverythingSmartHome