EverythingSmartHome/everything-presence-lite #149: EPL with SEN0395 - how to get full list of HA entities

svente08, 2024-08-23:
Hi guys,

I have an EP light with the SEN0395 and it basically work but the crucial entities are missing in home assisant. I tried many things to import config from ld2450-common.yaml without success. 

How can I get all entities available in HA like I have when running EPL with default sensor ld2450? Talking about the target/zone stuff. 

Cheers
Sven

EverythingSmartHome, 2024-08-23:
The SEN0395 doesn't support target zones or tracking info like the LD2450 does so that's why it doesn't show - both sensors have different options and entities

closed 2024-08-23 by EverythingSmartHome

svente08, 2024-08-23:
Maybe I didn't make myself clear enough - sorry.

I would like to use zones/distance - like the EP1 with SEN0395 does. 

Please dont get me wrong - I really appreciate the project - but current ha entity set for EPL with SEN0395 is nearly useless.

EverythingSmartHome, 2024-08-23:
I think calling nearly useless is a pretty big misrepresentation, the beta features the EP1 has for the SEN0395 doesn't add very much functionality at all. It's power is in how good it is at static presence.

Anyways I'm still unsure as to which your referring to, because you said: "How can I get all entities available in HA like I have when running EPL with default sensor ld2450? Talking about the target/zone stuff" - which sounds like you are referring to the zones 1-4 and target 1-3 that is available in the LD2450?

svente08, 2024-08-23:
yep, You got me right. - but I need more than 6 meter range - the way to my house is almost 9 meter --> SEN0395

Actually, I could live with a simple distance measurement with the EPL and SEN0395.

EverythingSmartHome, 2024-08-23:
Got it - in this instance we currently don't have the beta features available on the EP1 implemented in the Lite yet for the SEN0395. This could be an option in the future but not a priority currently.

You could try to build a custom config based on the EP1 beta which is available here:https://github.com/EverythingSmartHome/everything-presence-one/blob/main/common/everything-presence-one-sen0395-beta.yaml