EverythingSmartHome/everything-presence-lite #150: Zones clarifications: overlapping and spikes

Draghmar, 2024-08-26:
Documentation states that zones shouldn't overlap. But should they have gap between them? The min possible is 10mm which looks small but then again that could be enough for some calculations to go wrong. Setting boundaries to the same value seems like borderline case of overlapping though, isn't not?

I've seen that there are coordinates spikes from time to time. Enough to make me wonder if that could be one of the culprit of some weird behaviours. Zones have min/max and if coordinate goes outside of that when spike happens then...well what then? :) Is there any smoothing being done in calculations to prevent spikes?

As a reference latest testing with three zones set with values (with max distance set to 400cm):
- Zone 1, X: -1300/350, Y: 30/930;
- Zone 2, X: -1100/50, Y: 940/1440;
- Zone 3, X: -1550/600, Y: 1450/2850

Image below has some interesting parts, with two highlighted by me. The first one is thin line where some weird drop on X/Y happened. The second is quite huge area where there were some smaller spikes and targets count going down to zero even though target active stays on, and zone occupancy going on/off even though main occupancy stayed on.
[image]

diedvdyk, 2024-08-29:
In this girhub for the LD2450 https://github.com/TillFleisch/ESPHome-HLK-LD2450 is stated that zones may overlap.
I have 3 EPL's with overlapping zones and dont give any problem..

EverythingSmartHome, 2024-10-08:
I believe this isn't an issue but do let me know if still a problem

closed 2024-10-08 by EverythingSmartHome