EverythingSmartHome/everything-presence-lite #151: Unable to integrate in to Home assistant

gazzaman22, 2024-08-29:
My EPLite failed yesterday so I reset the power  plugged in to my PC and tried to readopt it but Home assistant only see a ESPhome device now , any ideas or do I need to RMA it
Regards Garry

EverythingSmartHome, 2024-08-29:
Can you share a screenshot of what your seeing? It sounds like you might have installed a blank firmware on it, if so please follow the factory reset section on the update page in the user guide

gazzaman22, 2024-08-29:
[image]
[image]
[image]
[image]
[image]
[image]

Hi I did follow the guide and this is what I get

EverythingSmartHome, 2024-08-29:
Thanks for the screenshots, that's not the guide. The guide is located here: https://everythingsmarthome.github.io/everything-presence-lite/updating.html#manual-setup-and-factory-resetting

The method you are doing installs a blank ESPHome firmware to the device. Follow the steps above to install the Lite firmware to resolve the issues

gazzaman22, 2024-08-29:
Hi you asked for screen shots of what I was seeing, I did use that guide and I cant get the firmware to install, my Home assistant does have Bluetooth but The EPlite isn't seeing it

EverythingSmartHome, 2024-08-29:
> Hi you asked for screen shots of what I was seeing, I did use that guide and I cant get the firmware to install, my Home assistant does have Bluetooth but The EPlite isn't seeing it

I think you are possibly getting confused, the screenshots you sent were not from the user guide. The link I sent above is the guide, you do not need to click any other links or navigate to any other page other than the one I posted above. Are you clicking the big home assistant button on the page above as per the instructions? Then select the LD2450 sensor and then select either Bluetooth or no bluetooth

gazzaman22, 2024-08-29:
Ok finally got it working, What was confusing me was the reference to the connect button , and once you pointed it out that its the home assistant icon that made more sense , also the button only seems to work for me 50% of the time and that's with Chome with no adds on any it works thanks for your help

closed 2024-08-29 by EverythingSmartHome