EverythingSmartHome/everything-presence-lite #155: mmWave updates stop after some time

thexperiments, 2024-09-03:
Currently after 1-3 Days my mmWave data seems to be stopping.
There is no update of the positions and occupation anymore. The Light sensor seems to still properly update.

What I tried:
* Restart mmWave sensor --> Some old detection data get's flushed but still no updates
* Factory reset mmWave sensor --> Similar to above, stream not restored
* Restart ESP --> weirdly even this doesn't fix it
* Unplug the powere and power it up again --> This fixes it but after some time it hangs again

My suspicion is that the mmWave sensor hangs up in a state where the reboot command is not working anymore

The logs don't show anything suspicious other than that the stream of serial data seems to stop and not recover.

I already tried to update the firmware on the mmWave sensor but it says it already is the latest one.

Is the power source for the mmWave sensor switchable via a GPIO? Sadly the sensor doesn't seem to have a reset pin.

EverythingSmartHome, 2024-09-03:
This is a known issue with older version of the mmWave firmware - can you please download the HLKRadarTool app on iOS or Android. When you open the app it will display the sensor it detects over bluetooth. Select the sensor and it should let you update, select 2.04 as the version. The app is quite buggy and it can take a few attempts to get it to go through but should eventually work.

Once updated, this issue should be resolved

thexperiments, 2024-09-03:
Ok didn't upgrade to 2.x as it is still marked as beta.

Yes app is very buggy and crashes on iOS upon connecting to the sensor but luckily I have an Android device available.

Still with some HW revision the option for cutting power with an GPIO would be a great recovery option.

I will update the sensor and check again

EverythingSmartHome, 2024-09-03:
It's marked as beta but not really - it's the firmware we ship all Lites with currently if that helps. Yeah I found the app works a bit better on Android but can be pretty tempremental

thexperiments, 2024-09-03:
Ok sure, let's see, upgrade worked fine.

Is this documented somewhere in a FAQ and I just didn't see it or is it just a rare issue as most ep-lites were shipped with a newer sensor firmware?

EverythingSmartHome, 2024-09-03:
It's something we've had a handful of reports of via email the last week or so and not yet added to the docs. If you have some time, a PR would be very welcome for the docs but if not I have a bunch of stuff to add soon.

And yes I think it would be a less common issue as only the first batch of Lites I believe shipped with that firmware, all of the ones since should be on the newer 2.04 from factory

EverythingSmartHome, 2024-09-03:
Thanks, just a tiny change to make

thexperiments, 2024-09-03:
fixed

EverythingSmartHome, 2024-09-03:
Thanks very much!

Let me know when your able to confirm issue is resolved

pdecker414, 2024-09-19:
I have one of those ld2450 in my Everything Presence Lite and have this issue as well. the mmWave goes offline at random times and only a power cycle fixes it. I have tried everything to be able to use HLKRadarTool and RadarTool but neither can connect since Bluetooth switch is not available in ESPHome Device settings and I am not able to find out what firmware is installed in the mmWave sensor. Any help would be appreciated
[image]

EverythingSmartHome, 2024-09-24:
> I have one of those ld2450 in my Everything Presence Lite and have this issue as well. the mmWave goes offline at random times and only a power cycle fixes it. I have tried everything to be able to use HLKRadarTool and RadarTool but neither can connect since Bluetooth switch is not available in ESPHome Device settings and I am not able to find out what firmware is installed in the mmWave sensor. Any help would be appreciated ![Screenshot 2024-09-18 190307](https://private-user-images.githubusercontent.com/114041494/368814639-3953a445-94b9-4833-a177-d4802d85b281.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNzczNDQsIm5iZiI6MTcyNzE3NzA0NCwicGF0aCI6Ii8xMTQwNDE0OTQvMzY4ODE0NjM5LTM5NTNhNDQ1LTk0YjktNDgzMy1hMTc3LWQ0ODAyZDg1YjI4MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNFQxMTI0MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNzFmODBiM2NhYzYyZjhmMDI4NDg1NTkyYWYwODY5ZDc5NmI1MTBkNzUyNGY4MDQxNzI5MjZkOGFkMGQ4ZmMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.HylfPcDUDfEaVpGYjH_PorJ53ZMPoqNEfuHzFQUE5i8)

This is a standard Home Assistant feature, you click the entity (mmWave Bluetooth), then hit the cog and hit enable to turn that entity on.

pdecker414, 2024-09-24:
Much thanks. I was racking my brain on how to do it.

Kummallinen, 2024-10-05:
For anyone else has this problem as sees HLKRadarTool crashing constantly on iOS, use "RadarTools" instead.

closed 2024-10-08 by EverythingSmartHome