EverythingSmartHome/everything-presence-lite #157: Feature request - angle of installation

simonlirwin, 2024-09-04:
The ideal placement for any sensor is the corner of a room to get the best coverage of corners of the room. 
However, all the zones are set up as "rectangles" in parallel with the sensor. If the sensor is placed at an angle all the zones would currently also be at an angle to the room, creating a lot of difficulty. So, the optimal zone setup means suboptimal room coverage. 
If we could set an angle of installation e.g. 45°, the x and y coordinates could be translated using trigonometry. Then all the zones can be aligned correctly with the room while also totally eliminating the blind corners.
This is something I don't think anybody else has done yet, and could give you a USP.

LarsStegman, 2024-09-28:
Yeah, this would be awesome!

EverythingSmartHome, 2024-10-08:
Now been added as of version 1.1.0

closed 2024-10-08 by EverythingSmartHome