EverythingSmartHome/everything-presence-lite #159: Devices stopped working after software update

clintebbesen, 2024-09-07:
Hi,

I updated most of my lite's a few days ago and after doing so all of the ones I updated all stopped working. They showed up in the  home assistant updates list, so I clicked update, going from 2024.8.0 to 2024.8.3, immediately stoping working when doing so. Below is one that I haven't updated yet showing what I did to do the update.

[image]

The lites all are not responding, and they have a red flashing light on the unit. home assistant shows them as unresponsive.

I just realised they do show up in ESPHome, looking like new devices?
[image]

If I click adopt will that mean they are being setup as new devices, and I lose everything I have done with them? I really don't want to lose the values I have entered into the zones and various other settings on each, naming, setup with automations, etc It took literally days (I am not kidding, I have spend around 5-6 nights, each 5-7 hours) to configure and fine tune the zones around the house and troubleshoot issues and work out workarounds by tweaking the zones and settings. this will be extremely frustrating if all of that is lost.

I have one left that I haven't updated yet, should I avoid the update, or do it in some other way?

EverythingSmartHome, 2024-09-08:
Are the original configs still in ESPHome for them?

clintebbesen, 2024-09-08:
I think so, but that wouldn't have the configuration thats stored on the device like zones would it? 

It would be great to get these working again, ive gone from them intermittently working to nothing at all. After 4 or 5 days of not having them, I really miss having them even intermittently working.

[image]
[image]

EverythingSmartHome, 2024-09-09:
It's most likely because you edited the name of the device in ESPHome instead of Home Assistant (that's my theory). Do not adopt the ESPHome config, but instead try and hit install on the original config and install it. You may need to do it over USB if wireless fails

clintebbesen, 2024-09-09:
Yes that’s because when I received them they all had the identical name, Id, host name etc so I couldn’t add them to esphome as they were identical and already existed after I added the first one.

EverythingSmartHome, 2024-09-09:
Try to flash the original config and see if it works, if not grab some logs

clintebbesen, 2024-09-09:
Finally got it installed via usb. Has to climb up a ladder and try pry the case off and after pushing the buttons on the pcb it installed via usb. 

It works in home assistant again and has the same config as before. 

What do I need to do in future to prevent all this happening again?