EverythingSmartHome/everything-presence-lite #162: How to enable bluetooth proxy?

NylonDiamond, 2024-09-08:
I'm unable to see the everything-presence-lite in the HA Bluetooth integration. I want to use the Bluetooth in the device as a proxy as well as use the mm wave sensor. Is this possible?

EverythingSmartHome, 2024-09-08:
You don't have to do anything to enable it, if you flash the bluetooth proxy firmware and add it to Home Assistant, it's now working :)

closed 2024-09-08 by EverythingSmartHome

NylonDiamond, 2024-09-08:
Shouldn't it show in the bluetooth integration? I have 3 of the EP1 lite's in the house flashed with the Bluetooth proxy enabled firmware and I don't see the Bluetooth proxies anywhere in HA...
[image]

EverythingSmartHome, 2024-09-09:
No the Bluetooth integration is only for a USB or built in Bluetooth. Bluetooth proxies just work automatically through the ESPHome integration and do not need an additional integration