EverythingSmartHome/everything-presence-lite #164: Highly variable illuminance

V4ler1an, 2024-09-12:
I raised [this issue](https://github.com/EverythingSmartHome/everything-presence-lite/issues/148) previously and was asked to update all the firmware, which i've done. Since then i've continued to see strange readings which has made using illuminance in my lighting automations impossible.

One example, my lounge and kitchen/diner are next to each other and i have the door open so I can see the light levels of both. Today they looked similar and the readings are this.
[image]

Then 30 minutes later, the light levels are still all pretty much unchanged (to my eyes) but these are now the reported levels:
[image]

Today is full cloud, there were no periods of direct sun then cloud. It's been very consistent light levels.
Why are these readings so erratic? When I first received the EPLs they all read very similar and much lower readings, typically 0 -> 50 or 60 ish. My automations with illuminance were working great. Now i've had to disable them.

[image]

[image]

sfranky, 2024-09-15:
Same problem here, values are highly variable!

sfranky, 2024-09-16:
Is this maybe related to [issue 140](https://github.com/EverythingSmartHome/everything-presence-lite/issues/140) ?

V4ler1an, 2024-09-16:
Unquestionably. My problems also began with updating past the 1.0.6 firmware. I can see from https://github.com/EverythingSmartHome/everything-presence-lite/issues/140 they are making progress on finding a solution. Rather than downgrade I will wait for an update.

EverythingSmartHome, 2024-09-16:
Please update to 1.0.10 which was just pushed, it reverts the changes to the light sensor. I will close this one as it's a duplicate of https://github.com/EverythingSmartHome/everything-presence-lite/issues/140 but hopefully the new version should resolve

closed 2024-09-16 by EverythingSmartHome

V4ler1an, 2024-09-16:
Great, thank you.

V4ler1an, 2024-09-21:
> Please update to 1.0.10 which was just pushed, it reverts the changes to the light sensor. I will close this one as it's a duplicate of #140 but hopefully the new version should resolve

I'm not seeing this firmware update in HA like the previous. Is there a delay in pushing the update out? Or should I do the web based update?