EverythingSmartHome/everything-presence-lite #165: Add a Z dimension for detection zone

dverb20, 2024-09-12:
I've noticed my kitchen detection does on when my robot vacuum cleans which makes sense, however, I'd like to find a way to not have my sensor detect things that are only a few inches off the ground. I can see how this could be challenging given you might not always know the height or angle of installation, but having the ability to lower the detection zone even more would really help improve detection accuracy. While there is a 'target angle' entity, I've noticed my target angle is almost always consistent regardless if it's myself or the vacuum in the kitchen, I suppose I could flip the sensor over? Anyway, food for thought!

OssariumSolferinensis, 2024-09-28:
I've positioned my EPL at the back of a shelf and placed a book flat in front of it. This blocks the lower part of the detection range so the cats won't trigger the light / music / etc when they walk around the bedroom or sleep on the bed. Not the most elegant solution but it works.
I am restricting the field of vision of my PIR sensors with masking tape but haven't tried wrapping tin foil around the bottom half of the EPL  :-D

EverythingSmartHome, 2024-10-08:
Unfortuantely the sensor itself does not output a Z axis and so there is no way to add this. As mentioned you can hack it by rotating the device, however you then lose another dimension so not ideal

closed 2024-10-08 by EverythingSmartHome