EverythingSmartHome/everything-presence-lite #166: Install firmware problem

perseus177, 2024-09-16:
Hi, 
Please help 

I bought Everything presence sensor lite. 
By mistake, I erased ESP board. 
I want to install the firmware back. 

I install it via web ESP Home. 
In Home Assistant in addon ESP Home I changed substitutions and package and click to install 

```
substitutions:
  name: everything-presence-lite-f1bcf8
  friendly_name: Everything Presence Lite f1bcf8
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

http_request:
  verify_ssl: false  
```

Result from Installation 

[... 267 lines cut ...]
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-rgb-led.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/libb64/cencode.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/main.cpp.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/stdlib_noniso.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/everything-presence-lite-f1bcf8/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/everything-presence-lite-f1bcf8/libFrameworkArduino.a
Indexing .pioenvs/everything-presence-lite-f1bcf8/libFrameworkArduino.a
Linking .pioenvs/everything-presence-lite-f1bcf8/firmware.elf
RAM:   [==        ]  17.1% (used 55984 bytes from 327680 bytes)
Flash: [==========]  102.8% (used 1885533 bytes from 1835008 bytes)
Error: The program size (1885533 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 800.07 seconds =========================
```

EverythingSmartHome, 2024-09-16:
Please follow the factory reset section in the user guide which will help you to reinstall the firmware:

https://everythingsmarthome.github.io/everything-presence-lite/updating.html#manual-setup-and-factory-resetting

perseus177, 2024-09-16:
@EverythingSmartHome 
I checked it but I am lost 

Here you can install the latest [ESPHome](https://esphome.io/) firmware on the Everything Presence Lite board for direct integration with [Home Assistant](https://home-assistant.io/)

I need more detailed information 
What I should do when I open page https://esphome.io/ ?

perseus177, 2024-09-16:
I tried it again via ESP Home Addon in Home Assistant 

```
substitutions:
  name: everything-presence-lite-f1bcf8
  friendly_name: Everything Presence Lite f1bcf8
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false

http_request:
  verify_ssl: false  
```

Result 

```
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/esphome-web-f1bcf8.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-lite-f1bcf8 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Installing ESP-IDF's Python dependencies
Collecting urllib3<2
  Downloading urllib3-1.26.20-py2.py3-none-any.whl (144 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 144.2/144.2 kB 3.3 MB/s eta 0:00:00
Collecting cryptography<35.0.0,>=2.1.4
  Downloading cryptography-3.4.8.tar.gz (546 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 546.9/546.9 kB 9.6 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: still running...
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting future>=0.18.3
  Downloading future-1.0.0-py3-none-any.whl (491 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 491.3/491.3 kB 9.8 MB/s eta 0:00:00
Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.8/61.8 kB 13.5 MB/s eta 0:00:00
Collecting kconfiglib~=13.7.1
  Downloading kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 145.7/145.7 kB 11.4 MB/s eta 0:00:00
Collecting idf-component-manager~=1.0
  Downloading idf_component_manager-1.5.3-py2.py3-none-any.whl (161 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 161.3/161.3 kB 11.9 MB/s eta 0:00:00
Collecting cffi>=1.12
  Downloading cffi-1.17.1.tar.gz (516 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 516.6/516.6 kB 11.3 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting packaging
[... 102 lines cut ...]

  It fails with the following output:

    Change Dir: /data/build/everything-presence-lite-f1bcf8/.pioenvs/everything-presence-lite-f1bcf8/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_22da5 && [1/2] Building C object CMakeFiles/cmTC_22da5.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_22da5
    FAILED: cmTC_22da5 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_22da5.dir/testCCompiler.c.obj  -o cmTC_22da5   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:381 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 854.86 seconds =========================

```

EverythingSmartHome, 2024-09-16:
> @EverythingSmartHome I checked it but I am lost
> 
> Here you can install the latest [ESPHome](https://esphome.io/) firmware on the Everything Presence Lite board for direct integration with [Home Assistant](https://home-assistant.io/)
> 
> I need more detailed information What should I do when I open page https://esphome.io/ ?

No, you just follow the instructions on screen. No need to visit any other page or click any links. Click the giant Home Assistant button on the screen and follow the steps

perseus177, 2024-09-16:
Hi 
@EverythingSmartHome 
Thank you - I didn't think to click on that icon

Summary 
1. Installation Not working via Home Assistant Addon, Supervised  
2. In the meantime I successfully installed it via Windows - with command line 
3. The best and faster solution was to install it via the web, but I was blind :D 
https://everythingsmarthome.github.io/everything-presence-lite/updating.html#manual-setup-and-factory-resetting
[image]

closed 2024-09-16 by perseus177