EverythingSmartHome/everything-presence-lite #169: Zone occupancy state issue?

vague666, 2024-09-16:
Not sure what is going on so trying to add all info I can and hope somebody can tell me if it is me doing something wrong or what.
I have the EPL at the end of my hallway and I control the hallway lights with it. The issue is that the lights turn off while I stand in front of it(tying my shoelaces, putting on jacket etc)
Homeassistant automation is waiting for a trigger for EPL to turn occupancy off before continuing the run, and this is where something goes wrong. I can see in state history the state being constantly on but HA still continues the run and the lights turn off. And the automation is not restarted until I leave the EPL zone
[image]
[image]
[automation.txt](https://github.com/user-attachments/files/17019793/automation.txt)
[history-2.txt](https://github.com/user-attachments/files/17019794/history-2.txt)

Draghmar, 2024-09-17:
My suggestion would to split this automation. You're saying that EPL Occupancy is on and doesn't change. So the issue most likely lays in the trigger logic. Try to make automation for turning lights on and second for turning off. So basically instead of having await for trigger you'd use automation trigger. For me at least that would tell me if the problem is related to that trigger.

vague666, 2024-09-17:
For whatever reason, this step fails, ie the automation appears to ignore it completely
(I changed the trigger to device instead of entity but that didn't help either)
```
wait_for_trigger:
  - type: not_occupied
    platform: device
    device_id: 38f2525300333131911163dd7a763dde
    entity_id: bb528ad1afc06a0abb9e7d3cf6c40091
    domain: binary_sensor
    for:
      hours: 0
      minutes: 0
      seconds: 10
timeout:
  hours: 0
  minutes: 0
  seconds: 0
  milliseconds: 0
```

The trace:
```
Executed: September 17, 2024 at 15:58:58
Result:

wait:
  remaining: 0
  trigger: null
```

vague666, 2024-09-17:
> My suggestion would to split this automation. You're saying that EPL Occupancy is on and doesn't change. So the issue most likely lays in the trigger logic. Try to make automation for turning lights [...]

I shouldn't have to do that. It's clear which step fails and it's the trigger wait. I don't know why though. The trace above is peculiar and I don't know if the firmware is missing something

vague666, 2024-09-17:
Seems to work now, I hope. For some reason I had to add the from value to the trigger too, ie state change from on to off even though the from state should be optional
In case anyone else is having issues:
```
wait_for_trigger:
  - platform: state
    entity_id:
      - binary_sensor.everything_presence_occupancy
    from: 'on'
    to: 'off'
    for:
      hours: 0
      minutes: 0
      seconds: 10
```

closed 2024-09-17 by vague666