EverythingSmartHome/everything-presence-lite #170: EPOL not booting properly after recent update

xkoldxx, 2024-09-20:
After a recent ESPHome update my EPOL has stop booting 

[12:04:50]abort() was called at PC 0x401be656 on core 0
[12:04:50]
[12:04:50]
[12:04:50]Backtrace: 0x400830ce:0x3ffe62f0 0x4009240d:0x3ffe6310 0x400985c2:0x3ffe6330 0x401be656:0x3ffe63a0 0x401bd203:0x3ffe63c0 0x401bd16a:0x3ffe63e0 0x400de631:0x3ffe6400 0x400fc399:0x3ffe6420 0x400fcd51:0x3ffe6440 0x40125dea:0x3ffe6460 0x40125d87:0x3ffe6480
[12:04:50]
[12:04:50]
[12:04:50]
[12:04:50]
[12:04:50]ELF file SHA256: d6a3b1bdac8b12b3
[12:04:50]
[12:04:50]Rebooting...

EverythingSmartHome, 2024-09-20:
Do you happen to have internet blocked on the device?

xkoldxx, 2024-09-20:
double checked my fw rules and no there is access to the internet on my IOT netowrk

EverythingSmartHome, 2024-09-20:
Can you try connecting it to the main SSID just to confirm? There is a bug in ESPHome that causes a boot loop like this when it can't reach GitHub

EverythingSmartHome, 2024-09-20:
Can you flash from the user guide again to check it works?

xkoldxx, 2024-09-20:
I have flashed it from there several times
- i even got a replacement sensor and still didnt work 

Heres my current cfg 

substitutions:
  name: "office-epol-sensor"
  friendly_name: Office EPOL Sensor
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: **************

wifi:
  ssid: !secret main_wifi_ssid
  password: !secret main_wifi_password

mdns:

EverythingSmartHome, 2024-09-20:
Which version are you flashing?

xkoldxx, 2024-09-20:
version of EPL or ESPHome?

EPL your site said 1.10 i think
HA ESPHome didnt state but im running latest

EverythingSmartHome, 2024-09-23:
No I mean Bluetooth or non bluetooth?

xkoldxx, 2024-09-24:
I run it with BLE but today I tried without but same error

James Kirkpatrick

EverythingSmartHome, 2024-10-08:
Can you post the startup logs from the device in full?

xkoldxx, 2024-10-29:
[
[office-epol-sensor_logs (1).txt](https://github.com/user-attachments/files/17550150/office-epol-sensor_logs.1.txt)
](url)

xkoldxx, 2024-10-29:
[office-epol-sensor_logs (1).txt](https://github.com/user-attachments/files/17550151/office-epol-sensor_logs.1.txt)

EverythingSmartHome, 2024-10-29:
It looks like internet is blocked in some way, there is an unfortunate bug in ESPHome that will cause a loop like this when internet is blocked and the device tries to check for updates on GitHub

xkoldxx, 2024-10-29:
What makes you say that this is the only esp device that doesn’t work I
onboarded 6 new esp devices today no issues

My opnsense firewall allows my IOT to iNet

James Kirkpatrick

EverythingSmartHome, 2024-10-29:
It only applies to devices that use the mew manufacturer updates components, most other devices don't.

It doesn't need to just be a firewall rule, it could be a misconfigured primary DNS server

t3chguy, 2025-01-09:
@xkoldxx any luck with this?

I had a rock solid EPL running on my iot network, then ordered another and a pair of CO2 sensors. Reflashed them and hitting the exact same log as submitted here

xkoldxx, 2025-01-11:
unfortunately not :/