EverythingSmartHome/everything-presence-lite #173: How do I update esphome version on Lite?

sddgit, 2024-09-24:
I've added a Lite directly to HA, and it's working fine. It's flashed to firmware 1.0.10. I've updated esphome itself to 2024.9.1. My other 2 esphome devices are showing as having updates available (from 2024.8.1 to 2024.9.1), but the Lite isn't showing the update. How do I get esphome firmware updated on it?

EverythingSmartHome, 2024-09-24:
What version of ESPHome does the lite show it's on?

sddgit, 2024-09-24:
2024.8.3. The other 2 esphome devices that I have are on 2024.8.1, and showing that updates to 2024.9.1 are available.

Thank you for the reply.

EverythingSmartHome, 2024-09-24:
Do you have the config adopted in ESPHome for the Lite? If so, you can just hit install and then wirelessly and it will update the version

sddgit, 2024-09-24:
Esphome shows this for the three devices. The other 2 devices are showing the Update option, but not the Lite…
[image]

EverythingSmartHome, 2024-09-24:
It can easily be because it's been compiled but not installed. Just hit install on the config

sddgit, 2024-09-24:
OK, I installed and it went through the compile and installation process. Still showed as on 2024.8.3. But then I noticed HA said it needed reconfiguring, and that asked for a new encryption key. I grabbed that from the esphome config and all was well. Now shows 2024.9.1 as being installed. Any ideas why I had to go through this process? Thanks so much for the support.

EverythingSmartHome, 2024-10-08:
Most likely when you first added it to Home Assistant, there is no encryption key. When you then add it to ESPHome, it automatically adds an encryption key for you (but isn't always obvious), meaning Home Assistant is still trying to communicate with the device without an encryption key while the device requires a key.

closed 2024-10-08 by EverythingSmartHome