EverythingSmartHome/everything-presence-lite #174: Occupancy sensor inconsistent with other sensors

tkadauke, 2024-09-24:
I have about 20 EPLs in my home, and virtually all of them have the same issue. Sooner or later, they enter a state where either

* "Occupancy" sensor is "Clear", but other sensors, such as "Target X Active" or "Zone X Occupancy" is on. I observe this when I enter the room, and my lighting automation is not reacting.
* "Occupancy" sensor is "Detected", but all other sensors imply that there's no detection. I observe this when nobody is in the room, but the lights keep going for hours.

Sometimes this can be fixed by waiting a few minutes, entering the room for a few minutes, exiting the room, and waiting for a few minutes. However, since this is happening at seemingly random times everywhere it the house, it makes lighting automations unusable.

Also, rebooting doesn't always fix it. I also updated the mmWave firmware to the latest version on each EPL, and power cycled most of them after the update (some I can't reach but this issue is happening for all of them). All other sensors (Target *, Zone *) seem to be consistent with what's happening in the room for the most part (I do sometimes observe curious delays, but everything snaps into place eventually. This is annoying during debugging, but no big deal in the long term). Rarely, I see a "Target 2 Active" or "Target 3 Active" stuck in "On" when nobody is in the room, so this makes it difficult to create a helper binary_sensor entity based on those as a workaround.

It seems that from a logic perspective, the "Occupancy" sensor should always reflect what the other sensors show.

I just updated to the latest version of ESPHome and wirelessly updated the firmware. I adopted the devices into the ESPHome addon for Home Assistant, but didn't modify the generated configs in any way.

EverythingSmartHome, 2024-09-24:
Thanks, can you please try and run the non Bluetooth version of the firmware as a test? I'm wondering if it's an issue where it becomes momentarily disconnected and the states become out of sync

tkadauke, 2024-09-27:
Thanks for the suggestion. I tested it right away and for approximately two days now on a subset of my presence sensors. I haven't seen them enter this pathological state since. So the no-ble workaround seems to work.

However, I'm using bluetooth proxy for controlling several devices throughout the house, so it'd be great to eventually get this to work. For the time being, I can use other ESPHome devices as dedicated bluetooth proxies. FWIW, my EP1s (which I use in the bathrooms because they have humidity sensors) haven't missed a single beat since I installed them :)

EverythingSmartHome, 2024-09-27:
Glad to hear it resolved.

It's unfortunately a by-product of having the Bluetooth radio and WiFi operating at the same time and something that would need to be worked on in ESPHome itself. It seems that some users do see momentarily disconnects when Bluetooth Proxy is active, and others have no issues at all and don't see the behaviour.

closed 2024-09-27 by EverythingSmartHome