EverythingSmartHome/everything-presence-lite #175: No firmware updates available for certain EPLs

Cyruz143, 2024-09-25:
I bought 2x EPLs when they first released with the new cases. HA picked them up and they've all worked flawlessly. I added a 3rd one today, it all went well but I noticed it A) was on a newer version of firmware out of the box and B) it was able to update itself to 1.0.10 (ESPHome 2024.9.1).

The two original ones are stuck on  Firmware: 1.0.3 (ESPHome 2024.5.1). I've tried removing them and re-adding them to HA and I still don't see a way to update the firmware via HA. I don't use any standalone ESPHome bits, just the HA integration. A few images for context below:

[image]

New EPL:
[image]
[image]

Older EPLs:
[image]
[image]

EverythingSmartHome, 2024-09-25:
Hello,

The "Everything Presence Lite Firmware" option you are seeing was added in a later release. Since you don't use the ESPHome add-on, I would suggest updating your existing units from the user guide here: https://everythingsmarthome.github.io/everything-presence-lite/updating.html#manual-setup-and-factory-resetting

You will need to do this over USB first, but after updated then future updates can be done wirelessly.

Cyruz143, 2024-09-25:
All done and super easy, shame Firefox isn't supported for web flashing but really neat process regardless.

Thanks!

closed 2024-09-25 by Cyruz143