EverythingSmartHome/everything-presence-lite #176: Disabling internet access for EPL sensors

peterbarbosa, 2024-09-26:
Hey all,

New user of Everything Presence Lite and finally had some time to test one out in my garage! I got it configured and working OK, but I realized when I update my firewall so the device can't access my WAN/Internet, the device becomes Unavailable after a few hours on HA.

I am using the factory firmware that it came pre-loaded with (purchases a few months ago). Are there any plans so this device can work without needing WAN/Internet access, or am I doing something wrong?

Thanks,
Peter
PS - keep up the great work. Looking forward to adding more sensors around my house as I get more comfortable!

EverythingSmartHome, 2024-10-01:
Hello! It's currently a bug in ESPHome that will cause a boot loop when it's unable to check for updates from the GitHub. Just to make it clear that its an unintentional bug and of course it should normally be completely fine to blocked internet. Hopefully ESPHome can resolve this soon.

mussawarakram, 2024-12-16:
Is there anywhere I can track this issue with ESPHome?

I am running into this problem myself.

t3chguy, 2025-01-08:
Looks like maybe https://github.com/esphome/issues/issues/6104