EverythingSmartHome/everything-presence-lite #177: Is there a known-issue with Unifi + EP Lite?

scrampker, 2024-09-27:
Recently either ESPHome, or my Unifi network was updated and now I'm getting ~/2 minute reboots. Unboxed new sensor just to check. Same issue. EPL was working fine for months.

I've seen various threads on this, related to ESPHome, but most of it is irrelevant. Seems like a lot of people were able to fix it by changing Wi-Fi power saving setting to 'none,' but that's no longer an option.

This happens even when doing a fresh install via USB from the ESH website through MS Edge using the web installer. I purposefully did not adopt into HASS ESPHome module -- just into HASS itself. It's about 2 minutes, and I see the same issue with both sensors. Interestingly enough the ESP32 baked into one of my Mi Smart Fan, running ESPHome does not have this issue.

The pings are wild against this sensor as well. Sometimes 13s, but the average is about 2000ms when it's not completely unresponsive.

**Network**

Network: Network v8.5.1
- Unifi UDM SE, Unifi OS: v4.0.20
- U6 Enterprise APs, v6.7.6

**Home Assistant**
Core 2024.9.3
Supervisor 2024.09.1
Operating System 13.1
Frontend 20240909.1

EverythingSmartHome, 2024-09-30:
Do you happen to have internet blocked to the device? There is currently a bug in ESPHome which will cause a boot loop when the device checks for updates and is unable to such as when internet is blocked. This is an unintentional bug of course and normally the device is completely fine to block internet on

scrampker, 2024-10-10:
Apologies for the delay on this response. Loss in the chaos. The unit should have access to the internet. The only things network-related that seem worth mentioning are:
1) I use a MAC filter on this Wi-Fi network, and this unit is whitelisted... otherwise I'd never be able to communicate to it.
2) The sensor is on my IoT VLAN/subnet, and my hass instance is on a separate.
3) I do have multiple APs in my house.

Both units appears to legit be rebooting. The red LED starts to blink, and pings drop if I have a console open running tests. Sometimes I get weird pauses even when it hasn't rebooted yet. Overall there's definitely something weird going on with the network connection.

[image]

EverythingSmartHome, 2024-10-10:
This is very similar to the bug I mention above. Make sure DNS servers handed out by DHCP are correct (DNS 1 needs to be working), and ensure no firewall rules on the subnet

scrampker, 2024-10-10:
Interesting. I just read that ESP32 cannot use BLE and Wi-Fi at the same time. So perhaps every time the EPL is doing a scan or talking to BLE devices, it drops Wi-Fi? I had never heard of this before. I guess does some sort of time-slicing since it shares the same antenna. 

I read that some people improve the situation by adding the following:
wifi:
  fast_connect: true

I'm flashing from the web again and seeing if I can get this thing working. I'll avoid adopting into hass. I'm wondering if somehow the default yaml that it uses is reprogramming it and causing issues.

scrampker, 2024-10-15:
I just wanted to say that there does seem to be some sort of issue with ESP devices, and Unifi APs. Found [this thread](https://community.home-assistant.io/t/if-you-run-a-unifi-access-point-and-your-esp32s-keep-going-offline/531982/21) and modified the minimum data rate for my 2.4GHz IoT Wi-Fi network, and now I'm seeing nearly 99% successful pings... much lower. Very seldom do I have 1000ms+ responses. Almost no drop-outs. I've been sitting watching my EPL for 15 minutes and it hasn't disconnected from hass, or blinked red even one time. The parameters are updating very rapidly. 

Essentially, night-and-day difference. Previously I would see sluggish updates in the hass UI. It's a crazy improvement.

[image]

scrampker, 2024-10-31:
2 weeks later and right back in the same boat. I just re-adjusted the density to lower it from 12 to 11mbps. I seem to be getting decently-consistent results again. Seems odd that this keeps happening, or that it would matter at all in the first place.

pmffromspace, 2024-11-19:
> Do you happen to have internet blocked to the device? There is currently a bug in ESPHome which will cause a boot loop when the device checks for updates and is unable to such as when internet is [...]

Is there more information available about this? @EverythingSmartHome 
Im also Running into trouble in offline setup, really sad because I just bought 3 Sensors and now I cant use them because they wont work offline. When will this be fixed? Is there an issue to track? I think a few issues of this repo relate to this..

Kind regards

scrampker, 2024-11-19:
About once a week I have to adjust the density. I bounce between 9, 11, and 12. What even more strange is that when the EPL gets 'stuck,' it still updates the lux sensor, but the proximity sensor appears to die. Once I update the density on the network, and the EPL recovers/reconnects, I magically get all sensor data. Definitely not normal. It's the only equipment I have that behaves that way, (including nearly 2 dozen ESP32 nodes I deployed in the last week which all work fine.

EverythingSmartHome, 2024-11-19:
> Is there more information available about this? @EverythingSmartHome Im also Running into trouble in offline setup, really sad because I just bought 3 Sensors and now I cant use them because they [...]

It would need be fixed in ESPHome unfortunately, it's beyond my dev skills. I can say that it's likely to do with how you are doing the block, as it doesn't happen every time and works as it should when you block in a certain way - I haven't figured out exactly which method of blocking causes it though

scrampker, 2024-11-19:
I'll give it a whirl.

> Did you try flashing the non Bluetooth firmware? It's a common issue when Bluetooth proxy is enabled

scrampker, 2024-11-20:
The non-proxy firmware is still working fine after half a day, which seems like a good sign. I'll report back in a few days. I would just say that perhaps if this is such a common problem, the proxy option should be secondary, and have a confirmation flag denoting the wide-spread issue with the ESP32 having to share airtime between Wi-Fi and BT.

My other dedicated BT proxies seem to work fine, but I imagine their workload is far lower than an EP sensor pulling double duty.

michaelheichler, 2024-11-26:
Greetings from Germany, 
I have the same problem like described here but the other way around. I am a bit confused, I have a Ruckus 650 AP which transmits on 2.4 ghz and 5 ghz. DFS Channels are used.

Now when I plug one of my four previously ordered units the Ruckus restarts or the Wifi will drop out blank. When I disconnect the unit everything is fine.

When I stand directly in front of the unit and it is on the device eg smartphone, notebook etc. The smartphone etc. Will loose wlan connection.

I have other Units like espresense on four esp32 tinker modules that dont produce no chaos. Also Aqara Fp2s....

I an glad when somebody has a idea what is going on here!