EverythingSmartHome/everything-presence-lite #178: Failing to initialize

terribleJim, 2024-09-30:
I'm trying to get the Everything Presence Lite (ELP) set up. I don't know if you are supposed to be able to connect directly via bluetooth thru an android device (tried via phone and tablet) but that doesn't work.. even tho it shows up as a bluetooth device (fresh out of the box).

I'm on gentoo linux and I've enabled the ch340 and 210 in the kernel and rebooted. My HomeAssistant (HA) is via a docker container on an odroid with no bluetooth. I've got a SLZB-06p7 zigbee2mqtt dongle connected via USB (don't have a POE switch yet.. planning to get one). On that dongle I've flashed it into bluetooth proxy, but haven't yet figured out how to enable that in my docker containers or yamls or wherever. 

When I plug the elp into a USB, a red LED flashes every second or so. Going thru the manual setup on the website listed under setup, I get to the point of 'Install' and then get the failed to initialize error. I've tried it by holding down the boot button as suggested in this video: https://www.youtube.com/watch?v=Rkvda7j3fNE where he shows that method. But he says the LED stays on without blinking. Mine keeps blinking. I've tried once to hold down the boot button while powering up the device and then the LED remained on without blinking. Went thru the install process but same results. Holding the boot button or not.

I have installed the HLKRadar Tool and it sees the presence sensor. I've tried using that to reinstall the latest firmware but it too fails with a "timeout".

Any advice?

EverythingSmartHome, 2024-09-30:
From what you describe it's almost certainly an issue with the USB driver or similar on Linux. I know that some distros (arch and others), the CH340 driver conflicts with some accessibility software that's installed by default called brltty:

https://github.com/arduino/arduino-ide/issues/1788

I'm not sure if this is installed by default on Gentoo but that's where I would start

For what it's worth, the Bluetooth method means you need to have a Bluetooth adapter installed on your Home Assistant server which you said you didn't have so that way won't work for you

terribleJim, 2024-09-30:
Thank you for the reply! 

I don't have brltty installed. I could try installing it but sounds like it is more a problem than otherwise.
```
$ emerge -s brltty
  
[ Results for search key : brltty ]
Searching...

*  acct-group/brltty
      Latest version available: 0-r2
      Latest version installed: [ Not Installed ]
      Size of files: 0 KiB
      Homepage:      
      Description:   System group: brltty
      License:       

*  acct-user/brltty
      Latest version available: 0-r2
      Latest version installed: [ Not Installed ]
      Size of files: 0 KiB
      Homepage:      
      Description:   user for brltty
      License:       

*  app-accessibility/brltty
      Latest version available: 6.6-r1
      Latest version installed: [ Not Installed ]
      Size of files: 3,599 KiB
      Homepage:      https://brltty.app/
      Description:   Daemon that provides access to the Linux/Unix console for a blind person
      License:       GPL-2 LGPL-2.1
```
I did try to manually install the EPL before enabling CH340 and CH210 in the kernel with the same error as with. I searched BRLTTY in the kernel (not knowing what else to look for) and "no matches".

So unless there's any other ideas, I guess I'll need to pursue getting my dongle going with bluetooth in HA. That seems very difficult from the web searches I've been doing - none so far have given a good recipe for my situation. But will keep looking.

EverythingSmartHome, 2024-09-30:
In Linux you also need to make sure that the user account has permission to access serial devices in the browser, that's the other typical issue users have on Linux

terribleJim, 2024-09-30:
Ok, I'll research that. I did `chmod a+rw /dev/ttyS0` and made sure I was a member of the group dialup. One thing that might be a cause? is there is no CH340 in my kernel.. just CH341. I assumed that would do it, but maybe not.  So my posts above aren't quite correct in that regard. I'll put it to the gentoo forums.

terribleJim, 2024-10-01:
This can be closed. I searched the ports initially and I swear it showed ttyS0 as connected. The EPL setup showed it as "paired". But when I searched dmesg again, and after disconnecting and connecting a few times to my USB port.. it clearly is showing ttyUSB0 as connected. In EPL setup, it showed both ttyS0 and ttyUSB0 as paired. But in choosing the ttyUSB0 and holding down the button after "install", I got it to work. Probably wasted some time and energy there. Apologies.

closed 2024-10-01 by terribleJim