EverythingSmartHome/everything-presence-lite #179: Phantom occupancy problem

V4ler1an, 2024-10-01:
I previously logged an issue similar to this with my kitchen EPL which was registering occupancy for no reason after I had left the kitchen. Following someone's suggestion, moving the Y starting coordinate of zone 1 +20mm away from the kitchen EPL seemed to sure the problem (I don't really understand why though...). 

Now I have my bedroom EPL also registering an occupany when I'm downstairs. Bizarrely, it does it only when I close my front door. The front door is using a Eufy sensor and the automation it runs has no bearing on the EPLs at all (not that I could even get an automation to artificially turn an EPLs occupancy on if I wanted to). When I watch the bedroom EPL's zone 1 occupancy sensor on my phone, it registers occupancy the instant I close my front door and stays occupied for around a minute or two. Then the zone 1 target count changes back to zero with the occupancy following about 2 or 3 seconds later.

I have no idea what to make of this, can anyone suggest anything to help me? There is no line of site to the bedroom EPL from my front door or reflective surfaces. It's up my stairs, through a door way and around a corner. The EPL is on the latest firmware. Any help appreciated.

EverythingSmartHome, 2024-10-01:
Is it directly above the door or similar? Also do you get any coordinates or targets tracked when it occurs?

V4ler1an, 2024-10-01:
It's at the top of the door to the right side. That said, i've just one moment ago gone from my lounge into the hall / lobby just inside my front door and had the bedroom occupancy trigger without even opening the front door. The only other sensors / automations in the lobby are a Eufy PIR motion sensor which turns on a matter over thread light bulb.  I also have the same model of Eufy motion sensor in my kitchen. 

Update: I just went back into the lobby and opening the front door triggered the bedroom again. This time I got a screen grab of the activity. All these values came up exactly together when it triggered, they were zero'd out before. I watched it happen. What do you make of this? By the way the illuminance is possibly because the bedroom lamp (matter bulb) came on with the occupancy as automated in HA.

[image]

V4ler1an, 2024-10-01:
Oh, and all those values remained static once they came up!

V4ler1an, 2024-10-01:
And again, totally reproduceable. All values pop up together and remain static. Notice how they are very similar. Oddly, it only does it as I close the front door. Not on opening. My front door automation does a bunch of tuff with various home security devices. None of which have ever given me any glithes or trouble when the automation runs. Alhough as I said, I gt the bedroom to trigger with just the lobby PIR sensor.

[image]

V4ler1an, 2024-10-01:
Wow, I think I might have found it. Across the bedroom from the EPL is a large double wardrobe with heavy glass fronted floor to ceiling sliding doors. They were roughly 20 degrees to the left of the EPL. I turned the EPL to face them and now the target registers as 3 degrees. The pressure change from the lobby and front doors must be causing the wardrobe to wobble very slightly. It has some give when I push it. I'll see if I can cut the wardrobe out using the zone setup. I think this must be it.  <rolls eyes>

V4ler1an, 2024-10-01:
Ok, it's solved. Apologies this was a tiny movement of those doors.

closed 2024-10-01 by V4ler1an

EverythingSmartHome, 2024-10-02:
Glad you figured it out. Just to clarify the point about the values being the same, as I'm sure you have figured out now, it's because it's actually tracking the source of movement which is coming from the same point all the time (the door in this case) so that's why the values always look very similar.