EverythingSmartHome/everything-presence-lite #183: EPL quickly becoming unavailable in HA

marinave, 2024-10-03:
Recently I purchased an Everything Presence Lite (EPL) sensor. I tried to connect it to my Home Assistant (HA), and although the initial installation went fine, the EPL soon became unavailable in HA. I’ve reinstalled the software several times, and while the sensor works for a short period, it quickly becomes unavailable again. HA still recognises that it is online, but none of the commands are functional. Any ideas how to solve it ? If it is a ESPHome issue any suggestion on how to get around it.
Many thanks

I am running HA on a Raspberry Pi 4 Model B with the following setup:
Core: 2024.9.3
Supervisor: 2024.09.1
Operating System: 13.1
Frontend: 20240909.1
ESPHome: 2024.9.2"

ELP firmware 1.0.10
LD2450 updated to latest version

EverythingSmartHome, 2024-10-03:
Hello, do you happen to have it blocked from internet access? There is currently an unintentional bug in ESPHome which will cause a loop when internet is blocked (and it is unable to check for updates from GitHub).

If you don't, can you please attach logs from USB?

EverythingSmartHome, 2024-10-03:
Thanks for the details. Those are the diagnostics but not the logs.

To get the logs, you can either use the ESPHome addon, or the update page in the user guide (select options for the firmware, hit connect, then logs)

marinave, 2024-10-05:
E (13446) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13446) task_wdt:  - loopTask (CPU 0)
E (13446) task_wdt: Tasks currently running:
E (13446) task_wdt: CPU 0: IDLE0
E (13446) task_wdt: CPU 1: IDLE1
E (13446) task_wdt: Aborting.

abort() was called at PC 0x40141240 on core 0

Backtrace: 0x400830ce:0x3ffbe410 0x4009240d:0x3ffbe430 0x400985c2:0x3ffbe450 0x40141240:0x3ffbe4c0 0x40083ac9:0x3ffbe4e0 0x401d3a73:0x3ffbc500 0x40141991:0x3ffbc520 0x400937e4:0x3ffbc540

ELF file SHA256: 1e424370a5c9025c

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6652
ho 0 tail 12 room 4
load:0x40078000,len:15052
load:0x40080400,len:3836
entry 0x4008069c
I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
I (29) boot: compile time 10:01:39
I (29) boot: Multicore bootloader
I (32) boot: chip revision: v3.1
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (73) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (88) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (103) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=48fe8h (298984) map
I (224) esp_image: segment 1: paddr=00059010 vaddr=3ffbdb60 size=050b4h ( 20660) load
I (233) esp_image: segment 2: paddr=0005e0cc vaddr=40080000 size=01f4ch (  8012) load
I (236) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=107074h (1077364) map
I (629) esp_image: segment 4: paddr=0016709c vaddr=40081f4c size=1c3e4h (115684) load
I (691) boot: Loaded app from partition at offset 0x10000
I (691) boot: Disabling RNG early entropy source...
I (703) cpu_start: Multicore app
I (703) cpu_start: Pro cpu up.
I (703) cpu_start: Starting app cpu, entry point is 0x400826b0
I (694) cpu_start: App cpu up.
I (723) cpu_start: Pro cpu start user code
I (723) cpu_start: cpu freq: 160000000
I (723) cpu_start: Application information:
I (728) cpu_start: Project name:     everything-presence-lite
I (734) cpu_start: App version:      2024.9.2
I (739) cpu_start: Compile time:     Oct  2 2024 10:00:40
I (745) cpu_start: ELF file SHA256:  1e424370a5c9025c...
I (751) cpu_start: ESP-IDF:          4.4.7
I (756) cpu_start: Min chip rev:     v0.0
I (761) cpu_start: Max chip rev:     v3.99 
I (766) cpu_start: Chip rev:         v3.1
I (771) heap_init: Initializing. RAM available for dynamic allocation:
I (778) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (784) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (790) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (796) heap_init: At 3FFCB8B8 len 00014748 (81 KiB): DRAM
I (802) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (808) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[... 780 lines cut ...]
I (694) cpu_start: App cpu up.
I (723) cpu_start: Pro cpu start user code
I (723) cpu_start: cpu freq: 160000000
I (723) cpu_start: Application information:
I (728) cpu_start: Project name:     everything-presence-lite
I (734) cpu_start: App version:      2024.9.2
I (739) cpu_start: Compile time:     Oct  2 2024 10:00:40
I (745) cpu_start: ELF file SHA256:  1e424370a5c9025c...
I (751) cpu_start: ESP-IDF:          4.4.7
I (756) cpu_start: Min chip rev:     v0.0
I (761) cpu_start: Max chip rev:     v3.99 
I (766) cpu_start: Chip rev:         v3.1
I (771) heap_init: Initializing. RAM available for dynamic allocation:
I (778) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (784) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (790) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (796) heap_init: At 3FFCB8B8 len 00014748 (81 KiB): DRAM
I (802) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (808) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (815) heap_init: At 4009E330 len 00001CD0 (7 KiB): IRAM
I (822) spi_flash: detected chip: gd
I (825) spi_flash: flash io: dio
W (829) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (844) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

larrybolt, 2024-10-05:
@marinave 
I was having a very similar issue (exactly the same, logs are slightly different:
```
E (11890) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11890) task_wdt:  - loopTask (CPU 0)
E (11890) task_wdt: Tasks currently running:
E (11890) task_wdt: CPU 0: wifi
E (11890) task_wdt: CPU 1: IDLE1
E (11890) task_wdt: Aborting./n
abort() was called at PC 0x40141240 on core 0/n

Backtrace: 0x400830ce:0x3ffbe410 0x4009240d:0x3ffbe430 0x400985c2:0x3ffbe450 0x40141240:0x3ffbe4c0 0x40083ac9:0x3ffbe4e0 0x4000bfed:0x3ffd97e0 0x40095402:0x3ffd97f0 0x40092ec5:0x3ffd9810 0x40143252:0x3ffd9850 0x401d3b59:0x3ffd9870 0x4018f85f:0x3ffd9890 0x401430bf:0x3ffd98b0 0x401a70e1:0x3ffd98d0 0x4009be1d:0x3ffd98f0 0x401d4112:0x3ffd9910 0x401d4179:0x3ffd9930 0x40099f46:0x3ffd9950/n
/n

ELF file SHA256: 1e424370a5c9025c/n
Rebooting...
```

I managed to figure out the issue was a faulty local dns server, the EPL is currently getting `1.1.1.1` (cloudflare's dns) and it works :)

marinave, 2024-10-06:
Thank you! I will try your solution.

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

docwho76, 2024-10-08:
So I recently observed an issue similar to this with my EPL units. It turns out that the first 2 DNS servers my router was handing out were down and the EPL never seemed to use the 3rd DNS host provided by DHCP. While the DNS hosts were down, the units basically constantly rebooted due to the perceived lack of internet.

EverythingSmartHome, 2024-10-08:
Yes this is currently a bug in ESPHome that will occur when manufacturer updates are enabled and internet is blocked or unreachable

marinave, 2024-10-09:
yes the issue was the dns server, once changed, the EPL was working perfectly. Thanks to all that helped.

closed 2024-10-09 by EverythingSmartHome