EverythingSmartHome/everything-presence-lite #186: EPL Lite unavailable after upgrading to 1.0.11

f18m, 2024-10-03:
HomeAssistant prompted me to run a firmware update from 1.0.10 to 1.0.11.
After I did that, the sensor was unavailable... I waited several minutes... nothing.

Then I went to ESPHome addon page in my HomeAssistant and clicked the "Install" action on the sensor. After re-installing from ESPHome dashboard the EPL is back functional.

**note**: my EPL has a CO2 sensor attached and thus I customized the ESPHome config following the guide

EverythingSmartHome, 2024-10-03:
That sensor is disabled by default to avoid this issue, I'm guessing you enabled it? The reason is that sensor downloads the update directly from GitHub, which overwrites the encryption key you have set in the ESPHome add-on. So if you have home assistant setup to connect with the encryption key (only possible when you use the ESPHome addon), the GitHub update will remove it and this HA won't be able to connect.

You should also keep that sensor disabled if your using the CO2 module as it will overwrite that code too.

Hope that helps!

closed 2024-10-03 by EverythingSmartHome

f18m, 2024-10-07:
ok, that makes sense, thanks!