EverythingSmartHome/everything-presence-lite #187: Everything Presence Lite not showing up in ESPHome

tomm1ed, 2024-10-05:
Configured using Bluetooth and is connected to Home Assistant. However has 1.0.6 installed and says no update available. In ESPHome not visible at all
[image]
How can I connect it to ESPHome so I can update to 1.0.11?

EverythingSmartHome, 2024-10-05:
Enable the hidden entity "everything presence Lite firmware" and you can update directly from GitHub without the ESPHome addon

EverythingSmartHome, 2024-10-05:
I think you would need to reboot (reboot button) the unit to get it to check after unhiding. Device needs to be able to reach the internet to check for updates of course

tomm1ed, 2024-10-05:
> I think you would need to reboot (reboot button) the unit to get it to check after unhiding. Device needs to be able to reach the internet to check for updates of course

Ah, no IoT device of mine gets to go on the internet so that would explain why no update is seen. 
I just succesfully updated through the Everything Presence website. Is there really no way to connect this device to ESPHome so I can update from there? Is the YAML you guys use not available anywhere? Tried using https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/everything-presence-lite-ha-ld2410.yaml but ESPHome then errors out on Validation:
[image]

Having to connect the damn thing to my computer of USB kind of defeats the purpose ;)

EverythingSmartHome, 2024-10-05:
Of course, your free to add to ESPHome. No need to do over USB. If you are using the docker version then it will show up automatically, if you have setup the container correctly. If it's not showing up automatically in ESPHome, then you have a misconfiguration 

If you want to add the YAML manually then yeah it's all on GitHub. It's a modular config that gets built automatically so you'll need to piece all the bits together starting with this one:

https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/everything-presence-lite-ha.yaml

(It's not LD2410 as per your previous link)

And then the packages referenced you will need to copy into one big file

tomm1ed, 2024-10-05:
> Of course, your free to add to ESPHome. No need to do over USB. If you are using the docker version then it will show up automatically, if you have setup the container correctly. If it's not showing [...]

Well I am running the docker version but it does not show up. As soon as I add a everything-presence-lite-9350e8 it does show as online but since I don't have a valid config that of course doesn't work ;)

[image]
(and yes I saw I copy-pasted the wrong link)

I think I succeeded in creating one big yaml, but have a new problem now. Since flashing to 1.0.11 the ESP appears to crash over and over with:

```
[C][api:025]: Setting up Home Assistant API server...
[D][button:010]: 'get_mmwave_firmware' Pressed.
[D][switch:012]: 'mmWave Configuration Mode' Turning ON.
[I][app:062]: setup() finished successfully!

abort() was called at PC 0x401b6196 on core 0

Backtrace: 0x400830ce:0x3ffe6610 0x4009240d:0x3ffe6630 0x400985c2:0x3ffe6650 0x401b6196:0x3ffe66c0 0x401b4d43:0x3ffe66e0 0x401b4caa:0x3ffe6700 0x400dde5d:0x3ffe6720 0x400f9d8d:0x3ffe6740 0x400fa745:0x3ffe6760 0x40123b7a:0x3ffe6780 0x40123b17:0x3ffe67a0

ELF file SHA256: 1e424370a5c9025c

Rebooting...
ets Jul 29 2019 12:21:46

```

EverythingSmartHome, 2024-10-05:
If it's not showing up then the docker container is misconfigured. You need mDNS to be able to pass through to the container.

You'll need to remove the http_request section (also present inside the ota block) since you have internet blocked - unintended bug in ESPHome currently

tomm1ed, 2024-10-06:
> If it's not showing up then the docker container is misconfigured. You need mDNS to be able to pass through to the container.
> 
> You'll need to remove the http_request section (also present inside the ota block) since you have internet blocked - unintended bug in ESPHome currently

Okay this has been a real goose chase. mDNS is apparently not working if you have a different subnet for your iOT devices and your ESPHOME docker, so it's not a 'misconfigured docker container' but it works as designed. Is my conclusion correct that EPL does not work when it's not in the same network as ESPhome and there is no way of fixing this? I have several other ESPhome devices that work flawlessly but there I have an actual complete YAML ;)

Question 2; is the **http_request** part the reason the EPL reboots constantly with the 
`abort() was called at PC 0x401b6196 on core 0`

line?

EverythingSmartHome, 2024-10-06:
From the edit history, glad you figured out it was a misconfiguration ;) to clarify, you can absolutely have devices in a separate VLAN and have mDNS work, it's what I do myself along with many others. The EP1 is no different in this regard and works the same as your other ESPHome devices. But seems like you figured that out in the end.

Yes there is a bug in ESPHome that when internet is blocked it will cause a boot loop like that when the device tries to check for updates from GitHub, which is what the http_request section does, so when you remove that and the associated config it should resolve it and stop boot looping.

tomm1ed, 2024-10-06:
> From the edit history, glad you figured out it was a misconfiguration ;) to clarify, you can absolutely have devices in a separate VLAN and have mDNS work, it's what I do myself along with many [...]
I'd call it a bug in the docker image and/or Ubuntu 22.04 but yeah ;)

> 
> Yes there is a bug in ESPHome that when internet is blocked it will cause a boot loop like that when the device tries to check for updates from GitHub, which is what the http_request section does, so when you remove that and the associated config it should resolve it and stop boot looping.

By allowing everythingsmarthome.github.io FQDN in firewall the boot loop is fixed. Now I can finally start playing with this thing :}

closed 2024-10-06 by tomm1ed