EverythingSmartHome/everything-presence-lite #188: can't find IP address

terribleJim, 2024-10-05:
Hello,
Hope I'm not wasting too much time but I can't find the IP address of the EPL. I plug the unit into my computer and go thru manual setup. I can connect and install firmware. When I click on setup wifi, or otherwise part of the update.. I get a list of SSIDs that must be from neighbors (who aren't that close but some of the SSID names make sense) but not really any of my own. I do see one that is mine but is names with a following _EXT which I don't know where that comes from. It accepts the password for the similarly named SSID (with the _EXT). But I go into my AP or router and it does not see the IP address. The last time I set the unit directly under the AP (couldn't find it away by my desktop) and typed in the SSID name iot for that vlan and it again said connected, but no evidence of IP address found.

My setup is using a OPNsense router from the ISP with a Grandstream 7660 AP connected to the router, that advertises our 3 or 4 vlans. See them fine on other wireless devices and they can connect. Perhaps there's a setting on the 7660 I'm missing for the vlan.. but looks good.

I'm far from integrating into my dockerized HA - lots to learn.. but first step probably would be locating the EPL.

I can go into the console of the EPL during setup and download logs.. which I've done. Can't find reference to IP addresses in them.. but wondering if there is a way to see it that way.. or see if it doesn't actually grab an address.

I've seen your advice to others on checking if "blocked". I don't believe I've ever blocked a device on purpose and would have to research to find that on my devices. Therefore I -don't think- that is an issue.. but...

appreciate all that you do!

terribleJim, 2024-10-05:
Ok, at a loss. I tried to connect to the chosen SSID and mistyped the password.. so wrong password.. when I hit connect it immediately said connected. But no sign in the router of a new client. I have tried several times since with sometimes not being able to connect and other times immediately connected without actually being connected. When it says "connected" (in the wifi dialog) it does it so quickly that it doesn't take the usual time for authenticating.. so I don't think I've ever connected it to wifi with success even though it reports otherwise.

EverythingSmartHome, 2024-10-06:
Hello! WiFi connectivity issues can sometimes be cause from a bad power supply, how are you powering it? Perhaps try configuring the SSID during the setup, then unplug and change to another know good quality power supply and seeing if it shows up then

terribleJim, 2024-10-06:
I'm plugging it into my desktop computer's USB port. It is an ASUS TUF GAMING B550M-PLUS WiFi II AMD, not that old motherboard. Newish powersupply for it. I also tried it plugging into the USB port on an ASUS Chromebook, though I expect if there would be power issues it would be that one. With that one being portable, I got it right under the AP but the EPL still didn't see the iot SSID. The desktop is a bit further from the AP.. the iot shows up as weak but still connectable. I guess I will order another AP to set up a mesh.. but that takes a few weeks for shipping.

Otherwise I have phone chargers.. have a 35 watt unit from google to try, and an old Samsung one. But no idea how to join the EPL not connected to a computer.. if that is possible I have hope. I've tried setting up some yaml's but I see so much contradictory info on the web.. not sure which way to go for docker compose. Need a good basic tutorial that would work - if that is a possibility. Seems in researching this I saw a post by you that indicated it was possible to get it working without necessarily being set up by plugging into a computer.

Thanks for help. I will give your suggestion another try.

terribleJim, 2024-10-11:
I let this sit since my last comment. I tried it a couple times in that duration, but figured it just wasn't going to work for me. Got a POE switch coming that I hope will boost the wireless AP more than my injector.. but doubtful.

Anyway, just tried it again.. doing nothing different except for the fact I forgot to use the USB cord plugged into my computer and instead grabbed the cord plugged into my phone charger. When I went to the setup webpage I noticed when hitting connect that the ttyUSB0 port wasn't listed - only serial ports. So I picked S0 and tried to connect. Of course it failed. Then I realized it wasn't plugged into the computer. Plugged it in and voila! For some reason, doing nothing different than all attempts before (except for the cord snafu), it discovered all the SSIDs I have configured and was able to join it to the iOt vlan. Router showed the IP address and I was able to join it to ESPHome in my HA install. So hopefully when I move it to its designated area, I can keep it working.

closed 2024-10-11 by terribleJim