EverythingSmartHome/everything-presence-lite #201: ESPHome adoption seems to rebuild the firmware and disable the CO2 sensor

SEWFlyers, 2024-10-08:
I just got 3 Lites yesterday and I am a HA newbie, so may be doing something wrong, but after installing the sensor and following the instructions for installing the right build, when I go into HA and try and configure the device to add it to my instance, I'm stuck in a loop that ends up with the device running without the CO2 sensor. 

When first adding the devices in HA, it asks for the encryption key, which it tells me to get through ESPHome. If I then adopt the device through the ESPHome interface so I can edit the YAML and get the key, it runs a script that seems to be compiling a new version, and when it succeeds at installing it, I can edit the config to see the key, but no longer see the CO2 sensor option in the config after it's been added. 

I have tried going through the process several times and at some point, have successfully gotten to a configured version with the CO2 enabled, so I know what it looks like when the sensor it active, but it's no longer working and I can't figure out how I was able to get there, nor why it didn't stay in that state. I've tried the BLE and non-BLE versions of the build and even at one point flashed it with the standard non-CO2 version to put it back in it's out-of-box state. 

Any thoughts on how to get it up and running with the CO2 sensor?  I'm connecting via the USB cable to my computer and have the drivers installed so I can flash the device directly. It says I have Firmware: 1.1.0 (ESPHome 2024.9.2) installed.

EverythingSmartHome, 2024-10-09:
Hello,

Sounds like because you are adopting it in ESPHome, it's overwriting the CO2 firmware with the base config.

I would recommend deleting from Home Assistant and the ESPHome add-on and starting again. Once you've done that, give HA a reboot to clear out potential cache.

Then go and flash the device with the CO2 firmware and re-add to Home Assistant. There should be no prompt for an encryption key as there isn't one set at this point. There is no need to open the ESPHome addon or adopt the config at this stage so recommend staying away from it till you get it added (we can add later if you like once you confirm you've added to HA)

limdul, 2024-10-09:
I got it working via ESP Home by replacing the package url in the config .yml

I'm running this currently:

`substitutions:
  name: "presence-bedroom"
  friendly_name: Presence Lite Bedroom
packages:
  EverythingSmartTechnology.Everything Presence Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha-co2.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret api_key
    

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: presence-bedroom.iot
`

SEWFlyers, 2024-10-09:
The core installation worked for me this afternoon after deleting everything and avoiding the ESPHome process. Thanks for that info, limdul, that makes sense and I'll try it again tomorrow. Now it's time for me to mess around with my Dashboards to take advantage of all the new info I'm getting from the sensors. ;-)

closed 2024-10-10 by EverythingSmartHome

SEWFlyers, 2024-10-10:
Quick question. Is there a way to edit the YAML without adopting it via ESPHome first? I'm still poking around, but I can't find the YAML for the three EPLs with CO2 sensors on them in the UI or via Text Edit anywhere, as I haven't adopted them in the ESPHome UI yet. 

I have three other EPLs without CO2 sensors and they are adopted, appear in ESPHome, and I can see their YAML files. But the three I haven't adopted seem invisible (or more likely, I just don't know where to look).

Trying to figure out if I need to adopt them, let ESPHome flash the wrong version, and then manually edit the YAML to give it the right version afterwards, or if I have missed an easier way.

EverythingSmartHome, 2024-10-10:
Counter question, what YAML edits do you need to make?

But yes when you adopt it you can press skip, you don't have to install it. Most people just don't notice the skip button

SEWFlyers, 2024-10-11:
Excellent question. Ha. Based on limdul's comment, I was going to swap out the line in the YAML to point to the correct build for the CO2 sensor, that way I can use ESPHome's update feature going forward after adopting the EPL. 

Being new to HA, I don't know if that's critical, but seemed like worth trying to figure out, that's all. Plus experimenting will let me learn a little more about YAML and how this whole system works. 

And I didn't know exactly how the skip button would work, but that makes total sense. I guess I just didn't trust it. Rookie mistake. 🤦‍♂️