EverythingSmartHome/everything-presence-lite #202: Request: implement a sensitivity setting.

SynoUser-NL, 2024-10-09:
Hi,

I've bought several EPL's which all appear to be working perfectly with Home Assistant. Thank you for a great project!
I would like to place a couple of those EPL's in rooms where pets are frequently present and roaming around (cats, dog). These small animals obviously also trigger an an occupancy detection, whereas I'm only interested in the bigger animal (human) presence.

Would it be possible to implement some kind of sensitivity setting in a firmware update so small (animal) movement can be filtered out?

Thank you!
DenW

EverythingSmartHome, 2024-10-09:
It's not possible unfortunately since this would need to be implemented in the actual firmware on the mmwave sensor itself which is produced by HiLink. I've actually requested this several times over the last year from them and they said they can do it (for a cost) however it wouldn't give the desired results due to the algorithm and recommended against doing it.

closed 2024-10-09 by EverythingSmartHome

SynoUser-NL, 2024-10-09:
Ok.. I see. That's too bad.

The information on your site says that a different sensor (LD2410C and others) would allow for configuration of sensitivity.  So I'd need to change the mmWave board and reload the firmware to support that board. Correct?

Is there any board you would recommend for discerning between humans and pets this way?

EverythingSmartHome, 2024-10-09:
Yes you can switch to a different unit like the SEN0609, SEN0395 or LD2410C for configurable sensitivity. IMO none of them will do what you want, there isn't really a sensor available currently that can differentiate between animals and humans accurately. Those sensors do have sensitivity options but it's more for tiny movements where as any pet is going to be large enough to trigger them

SynoUser-NL, 2024-10-09:
Hmm.. yes I thought that might be the case.
Ok, thanks for your support and insight.