EverythingSmartHome/everything-presence-lite #207: Change the until of measurements from MM to IN

Mikede78, 2024-10-09:
Check the screen shots. You can see the Control area is setting MM and the other area is setting IN. How do I change that?

[image]
[image]

EverythingSmartHome, 2024-10-10:
This is a Home Assistant setting. Click on the entity, hit the cog and then change unit of measurement. The default for all entities is mm, your HA might be changing it based on your general settings. I'd strongly advise keeping them all at mm

closed 2024-10-10 by EverythingSmartHome

Mikede78, 2024-10-10:
I can't find it. Is it on this screen?

[image: image.png]

Mikede78, 2024-10-10:
> <EverythingSmartHome/everything-presence-lite/issues/207/2404451923@
> github.com>

Nevermind I found it. Thank you for your help.