EverythingSmartHome/everything-presence-lite #211: ESPHome version 2024.10.0 breaks EPL

f18m, 2024-10-17:
I updated today ESPHome to its new version 2024.10.0.
After doing so, my EverythingPresenceLite stopped working. If I go to the ESPHome dashboard, click on the component and choose "Validate" or "Update", I get the following error:

```
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/everything-presence-lite-4f1304.yaml...
Failed config

i2c: [source /data/packages/0c1ca49c/common/everything-presence-lite-base.yaml:34]
  - id: bus_a
    sda: 21
    scl: 22
    scan: True
    sda_pullup_enabled: True
    scl_pullup_enabled: True
    frequency: 50000.0
  - 
    ID bus_a redefined! Check i2c->0->id.
    id: bus_a
    sda: 21
    scl: 22
    scan: True
    sda_pullup_enabled: True
```

my full config is:

```
substitutions:
  name: everything-presence-lite-4f1304
  friendly_name: Everything Presence Lite 4f1304
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha-no-ble.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: OOXVfkO8pPHY/+uwUzUgojQiueUdrkMDrm4+stjKbbw=

wifi:
  networks:
  - ssid: FASTWEB-270816
    password: !secret wifi_password
  - ssid: TendaP1_270816
    password: !secret wifi_password
  - ssid: TP-Link270816
    password: !secret wifi_password

i2c:
  - id: bus_a
    sda: 21
    scl: 22
    scan: true
  - id: bus_b
    sda: 26
    scl: 27
    scan: true

sensor:
  - id: !extend illuminance_sensor
    i2c_id: bus_a
  - platform: scd4x
    i2c_id: bus_b
    co2:
      name: "CO2"
```

EverythingSmartHome, 2024-10-17:
I would advise removing the extra config that was needed for the CO2 sensors previously, then change the package line to include the CO2 config (since we now have prebuilt files for CO2)

MikeDembek, 2024-10-17:
> I would advise removing the extra config that was needed for the CO2 sensors previously, then change the package line to include the CO2 config (since we now have prebuilt files for CO2). So the config would look like this:
> 
> ```
> substitutions:
>   name: everything-presence-lite-4f1304
>   friendly_name: Everything Presence Lite 4f1304
> packages:
>   EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha-no-ble-co2.yaml@main
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
> api:
>   encryption:
>     key: OOXVfkO8pPHY/+uwUzUgojQiueUdrkMDrm4+stjKbbw=
> 
> wifi:
>   networks:
>   - ssid: FASTWEB-270816
>     password: !secret wifi_password
>   - ssid: TendaP1_270816
>     password: !secret wifi_password
>   - ssid: TP-Link270816
>     password: !secret wifi_password
> ```

I had this issue too and that recommendation worked for me.  Thanks!

f18m, 2024-10-17:
Indeed this configuration worked for me as well, thanks!

closed 2024-10-17 by EverythingSmartHome