EverythingSmartHome/everything-presence-lite #212: Presence not being detected in zone

Lampornis, 2024-10-18:
I set up an EP Lite a couple days ago to manage lights in my kitchen.  To confine presence detection to the kitchen I set up a zone that covers the covers the kitchen but excludes other areas within the range of the EPL.  After tweaking the the size of zone 1 a bit I have this working fairly well except that periodically when a target is in zone 1 the EPL will detect occupancy but zone remains clear.  Thus the automations I have that depend on occupancy in zone 1 being detected do not work.  I have tried restarting Home Assistant and power cycling the EPL multiple times which has not resolved the issue.  I have been able to kick start it into working again occasionally by rebuilding zone 1.  Thoughts on what the problem might be?
[image]
[image]
[image]

Lampornis, 2024-10-18:
> Can you share what the Zone 1 beginning and end X and Y values are? They are cut off in the screenshot so just wanted to double check which one is which, I think I see the issue

Sure things:

Beginning X: -1820.0mm

Beginning Y: 310.0mm

End X: 2170.0mm

End Y: 5080.0mm

EverythingSmartHome, 2024-10-18:
Are you sure? Cause the screenshot has the begin Y value as 5080?

Lampornis, 2024-10-18:
Sorry, I flipped the starting and ending Y values.  Yes, 5080 is the start and 310 is the ending value.  X values are correct.

EverythingSmartHome, 2024-10-18:
That will be the issue, the start value needs to be the lower value and the end value is the higher one. That goes for the X values too where a minus number would go as the start X.

Did you you change the values around given by the add-on or did it add those?

Lampornis, 2024-10-18:
I did not enter those values.  The values came from the add-on.  Wondering if the order of the numbers is being determined from where I start drawing the zone (bottom left, top right, etc.)?  Can I fix this by just inverting the numbers or is it not that simple?

EverythingSmartHome, 2024-10-18:
Yes you can just invert them.

The documentation for using the tool does mention to start with the left corner first then draw to the bottom right corner which produces the correct results

Lampornis, 2024-10-18:
LOL, I'm sure it absolutely does mention it.

EverythingSmartHome, 2024-10-18:
It wasn't a dig or anything (not sure if you took it that way), just saying I was always aware of that limitation with the add-on (hopefully can improve it in the future so it doesn't matter) so that's why I added it to the docs.

Did it fix the issue btw?

Lampornis, 2024-10-18:
Oh no, never thought it was a dig.  Heck, I'm a Mac guy.  What do I know about reading documentation.  Anyway, I'll know by tomorrow if it is still an issue.  Fixed the values so we shall see.  We have a big discussion going on over in Shane Whatley's discord about your add-on and am sharing all this with several others who are diving in on the EPLs.  Love love love the EP1 and EPL BTW.

EverythingSmartHome, 2024-10-18:
Love to hear it :)

You can get me over on Discord too btw on the ESH server if there is further questions, there is a whole EPL and EP1 channel for discussion. 

I just added a docker version for the configurator on the off-chance anyone isn't using HA OS

Lampornis, 2024-10-18:
Spoke too soon.  Did not resolve the issue.  I'm going to manually redraw the zone the correct way and see if that makes a difference.

EverythingSmartHome, 2024-10-18:
I just noticed that in your above screenshot, the coordinates it has for the target are outside of the zone which is why it's not tracking it. You have the zone set to -1800 for the begining X value, but the target is at -2000 so it would be outside. Looks like you need to increase the zone to catch the edges

Lampornis, 2024-10-18:
Only a small portion of the zone is outside the detection range of the sensor (see attached photo).  It is a limitation of a rectangular zone if I want to cover the entire kitchen.  On one side the area outside the zone is behind a wall so it will never be entered.  On the other side is an area I do not want detection anyway.  Is it a problem to have small portions of a zone outsied the detection area of a sensor?  If so not sure zones will work for me in this case.  I did redraw the zones and as of right now all is working really well.  We shall see.  Thanks for all the help you have given me today.  I really appreciate it.

Cheers,

Don

￼

Lampornis, 2024-10-18:
Here is the photo that did not come across in my previous response.
[image]

Lampornis, 2024-10-18:
The other possibility, I suppose, is that the coordiates that listed do not jive with the image of the zone.  I will say that when I track targets in the kitchen the do stay withing the visual zone.

Draghmar, 2024-10-18:
If one of your borders is at the very edge you'd like to track, then you'll get detections outside. I had the same issue in my bathroom, where I've set three zones to make for different parts that I didn't want to track. But the issue is that when for example someone stops moving at the border but, lets say, move arms, then it most likely be interpreted as movement outside of the zone. In my case it was when someone would lean over washing machine.

Lampornis, 2024-10-18:
The issue I was having though was that any presence within the zone was not being detected.  

The other thing I should mention, which may or may not be important is that the sensor is rotated so that it is not reading perpendicular to the wall that it is mounted to.  It is rotated to point at the kitchen.  Not sure if this is important or not.

Draghmar, 2024-10-18:
> The issue I was having though was that any presence within the zone was not being detected.
> 

Ups, sorry. I read your zone description the other way around ;)

MenesesPT, 2024-10-19:
> The other thing I should mention, which may or may not be important is that the sensor is rotated so that it is not reading perpendicular to the wall that it is mounted to. It is rotated to point at the kitchen. Not sure if this is important or not.

It is relevant, if you changed the installation angle in the sensor device page, under configuration. Otherwise, it should not influence.
The current version of the configurator doesn't take into account the rotation set in the configuration, so the target and the zones could not be shown correctly. I've already done the coding that fixes this, but I hadn't time to fully test it yet, but I'll try to do so today :)

Lampornis, 2024-10-19:
> It is relevant, if you changed the installation angle in the sensor device page, under configuration. Otherwise, it should not influence. The current version of the configurator doesn't take into [...]

I did not change the installation angle.  Anyway, thus far today it is seems to be working well so hopefully we have resolved the issue.

bazcurtis, 2024-10-19:
I am so pleased I found the Zone Tool. It makes life so much easier. I draw my two zone to be roughly what I wanted and then updated the coordinates via the entities and watched zones changed. I am seeing this. I am in zone 2, but zone 1 is active. The target count is correct for zone 2 and so is the occupancy. Shouldn't zone 2 be active and 1 inactive?

[image]
[image]

EverythingSmartHome, 2024-10-19:
Perhaps I'm missing something but in the screenshot above zone 2 is active and zone 1 is inactive? Looks correct to me.

You have EPL Gym Zone 2 Occupancy as active and EPL Gym Zone 1 Occupancy inactive

Lampornis, 2024-10-19:
> I am so pleased I found the Zone Tool. It makes life so much easier. I draw my two zone to be roughly what I wanted and then updated the coordinates via the entities and watched zones changed. I am [...]
I agree with Lewis.  Based on your images it looks like it is working perfectly.  Are you perhaps confusing gym occupancy with gym zone occupancy?

bazcurtis, 2024-10-19:
I see what you mean. I thought Target 2 was in Zone 2, but that is the number of targets being detected, correct? There is no target 2? Sorry about that.

Lampornis, 2024-10-19:
> I see what you mean. I thought Target 2 was in Zone 2, but that is the number of targets being detected, correct? There is no target 2? Sorry about that.

That is correct.  Even so celebrate that it is working correctly!  That is the important thing.

Hoohm, 2024-10-21:
I have a similar issue, here is the mystery. The target 1 value sees me and does not let go, but the zone occupancy looses me even though I'm just sitting on my sofa. 
[image]

EverythingSmartHome, 2024-10-21:
> I have a similar issue, here is the mystery. The target 1 value sees me and does not let go, but the zone occupancy looses me even though I'm just sitting on my sofa.

Can you share the zone setup of the room? How big is the zone in comparison to the zone?

Hoohm, 2024-10-21:
Sure, I have two zones, one behind the other, I used the default x for both (-4000 to 4000) I did not check if it was relevant, in this scenario I'm sitting in the middle of the room which is in zone 1. Zone 1 has a y of 0 to 3600.

Zone 2 is behind zone 1, the only change is in y (3600 to 6000).

EverythingSmartHome, 2024-10-21:
How close is the sofa to the edge of the zone?

closed 2024-10-22 by Lampornis