EverythingSmartHome/everything-presence-lite #213: Installation Angle setting not showing up in DEvice Dashboard

FCalleja, 2024-10-19:
I was following the setup guide from the QR code and noticed my Everything Presence Lite is not showing the slider for Installation Angle seen there, even when clicking to see all entities, this is all I see:
[image]

My expectation was to be able to adjust the Installation Angle slider seen in the guide here: https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/home-assistant-entities.html 
[image]

The firmware shows as up to date, so I'm not sure if something else is needed to configure this? I need it because my sensor is in a corner.

Any thoughts on what the problem might be?

EverythingSmartHome, 2024-10-19:
Hello, do you have the ESPHome addon installed or no? If not, no need to install it I just need to know if you have it

FCalleja, 2024-10-19:
Hi there! I do have the ESPHome addon installed. I do have an update, though, when I woke up this morning there were update notifications for both the EPL firmware and the configurator add-on, and I applied them and the installation angle is now there :) 

[image]

Really happy with this sensor so far, thank you! closing the issue

closed 2024-10-19 by FCalleja