EverythingSmartHome/everything-presence-lite #214: Zone Configurator - Error Fetching Data

bazcurtis, 2024-10-19:
I have seeing this in the Zone Configurator. Any idea why that might be? It is working for my other EPL. I will reboot it later when I can a chance, but I thought I would mention it in case you have others report it.

Thanks so much for the Zone Configurator. It is so, so helpful.

[image]

bazcurtis, 2024-10-19:
I have just rebooted it and still get the same error.

MenesesPT, 2024-10-19:
I've faced the same problem, turns out I had disabled some entities in my HA. After I enabled them again it started working.

bazcurtis, 2024-10-20:
Can you remember which ones? This EPL only has one zone.

EverythingSmartHome, 2024-10-20:
You need to enable the other 3 zones (this is a home assistant feature) by clicking the entity, hitting the cog and then enable

bazcurtis, 2024-10-20:
OK, thanks. Do you have plans for the tool to support one zone only? On my other EPL I only have 2 zones active and it works on that one.

EverythingSmartHome, 2024-10-20:
It does support one zone only, you don't have to use the other zones just because the entities are enabled

closed 2024-10-25 by bazcurtis