EverythingSmartHome/everything-presence-lite #220: EP Lite became unavailable out of the blue

Lampornis, 2024-10-22:
When I got up this morning my kitchen lights automation tied to the EP Lite fired off as expected.  Then, 15 minutes later the EPL went unavailable.  Tried all the normal stuff to get it working again (rebooted HA, clean build file install in ESPHome, etc) but none of that has worked.  Tried to look at the logs but it appears there is some sort of communication issue that I do not know how to resolve.  I've attached illustrating the error messages I have recieved both with trying to get the logs and when I tried to do a clean file install.  Oddly, ESPHome did an update this morning and the EPL updated just fine (after it went unavailable).  ESPHome shows the EPL as being online.
[image]
[image]

EverythingSmartHome, 2024-10-22:
Not really possible to tell from these logs unfortunately as could be many things. Are you able to check logs over USB?

Lampornis, 2024-10-22:
Never mind.  Gosh, when I took my computer upstairs to plug it into the EPL to get the logs HA decided to inform me that I needed to re-enter the encryption key.  Would have been nice had they just told me before I spend the last hour and half trying to figure this out.

closed 2024-10-22 by EverythingSmartHome