EverythingSmartHome/everything-presence-lite #223: Update to 1.2.0 is failing.

Ba22a, 2024-10-22:
I ran the update but the device becomes unresponsive after the update finishes.

it is still available in ESPHome but no readings from the device.

if I choose to install from ESPHome it springs back to live but is running 1.1.1.

I am using everything-presence-lite-ha-co2.yaml@main as I have the co2 sensor.

thanks
Barry

EverythingSmartHome, 2024-10-22:
Which update are you running?

I'm almost certain this is a user issue (in the nicest way possible 😀), I'm guessing you have enabled the hidden entity that is "Everything Presence Lite Firmware"? The reason it's hidden by default is because if you have done extra steps of adding it to the ESPHome addon, then it will cause this issue because the ESPHome addon automatically adds an encryption key, where the pre-built firmwares do not have one. So updating through this entity removes the key and HA can't communicate anymore. There have been a few issues raised on this in the past, it's not a firmware issue

Ba22a, 2024-10-22:
Makes sense, so how do I upgrade? clearing the build files and hitting install isn’t upgrading either.

scotttimmins, 2024-10-23:
Going from information [here](https://community.home-assistant.io/t/esphome-addon-caches-yaml-depndency-from-github/657415) any external packages only update once every 24 hours.

If you're more technically minded you can delete the appropriate subfolder from esphome manually (stored in `.esphome/packages`). Of course, this is done at your own risk...

Ba22a, 2024-10-23:
Perfect, thanks.
Everyday is a school day. 24hr pointer is very useful. Thanks again. All sorted.

Taomyn, 2024-10-23:
@scotttimmins that's good to know, as finally I was able to update mine - I just wish I knew where the .esphome/packages folder was as I cannot find it anywhere via SSH on my OVA based HA server.

bharvey88, 2024-10-24:
> @scotttimmins that's good to know, as finally I was able to update mine - I just wish I knew where the .esphome/packages folder was as I cannot find it anywhere via SSH on my OVA based HA server.

`packages:
  ApolloAutomation.AIR-1:
    url: https://github.com/ApolloAutomation/AIR-1
    ref: main
    files: [Integrations/ESPHome/AIR-1.yaml]
    refresh: 1min`
    
    you can do something like this and get it down to 1min.

scotttimmins, 2024-10-24:
I run a docker esphome image and `.esphome` is in the config folder for me.
The `.` prefix can hide it in default listings but an `ls - a` shows it.

Taomyn, 2024-10-25:
> I run a docker esphome image and `.esphome` is in the config folder for me. The `.` prefix can hide it in default listings but an `ls - a` shows it.

Nope
[image]
[image]

scotttimmins, 2024-10-25:
That's your Home Assistant config folder - I run ESPHome as a separate container.

Maybe someone who runs it as an addon could give some guidance on where to look?

Taomyn, 2024-10-25:
> That's your Home Assistant config folder - I run ESPHome as a separate container.
> 
> Maybe someone who runs it as an addon could give some guidance on where to look?

I have ESPHome installed as an HA integration and as an HA add-on