EverythingSmartHome/everything-presence-lite #224: Auto Update seem to remove CO2 Sensor

Oliver3911, 2024-10-25:
Hi, 
I am using 2 EPLs one with CO2 Sensor and one vanilla.
I just added them to Home assistant, no ESPhome add-on.
The CO2 one has the firmware installed as written in the docs (1.1.1)
The 1.2.0 firmware which came in these days installed fine to both via the home assistant notification.
But now the CO2 Sensor ist not showing anymore.
I thing the Autoupdate installed the wronge firmware. But I don't know anything about ESPhome stuff so just my opinion.

What would be the best why to update the CO2 one?
Also could the Autoupdate be fixed?

Love these sensors, many thanks for the hard work!

Best regards

EverythingSmartHome, 2024-10-25:
Are you using the ESPHome addon for updates? If so there was a bug that was fixed a few days for this but you'll need to delete the config from the ESPHome addon and re-adopt.

You can also change the packages line to include "-co2" just before the .yaml:

`github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main`

Would become:

`github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha-co2.yaml@main`

Then save and install

EverythingSmartHome, 2024-10-25:
Ah apologies, this would affect the GitHub update also but was fixed at the same time. You would need to update via USB in this case and then updates going forward would apply OK

Oliver3911, 2024-10-25:
Okay update via usb worked but did reset the device. But thanks to the angle function setting up the zone was easy 😁
Let's hope I do not need to do this for the next firmware update.

Still many thanks and have a nice day

closed 2024-10-25 by EverythingSmartHome