EverythingSmartHome/everything-presence-lite #227: Wrong Units (mm vs in)

duckdude555, 2024-11-01:
Just set up my Everything Presence Lite (manually flashed newest firmware to work with Home Assistant) without issue - but when I try to use the Zone Configurator, I notice that the unit is reporting distances in inches despite claiming they are in millimeters. For example, I am sitting approx. 216in away from the unit now, but it is being reported as “216.4 mm”. I can’t seem to find a toggle to correct this… thank you!

[image]

EverythingSmartHome, 2024-11-01:
Hello! This is a Home Assistant setting and the Lite by default has it's units of measurements set to mm. If your Home Assistant has changed it to inches, you can click the entity, hit the cog and then select mm from units of measurement

closed 2024-11-01 by EverythingSmartHome

duckdude555, 2024-11-01:
Thank you! Can confirm this did the trick.