EverythingSmartHome/everything-presence-lite #228: Phantom target when I'm just beyond the edges of the sensor angle range

simonlirwin, 2024-11-02:
At the very edge of the angular range I'm getting the presence detected on the wrong side of the sensor. So if I'm standing at roughly 60-65 degrees, where the sensor has lost sight of me, the sensor starts showing me as being somewhere in the range of -45 to -65 degrees.
In the screenshot attached I show where the sensor is showing the target, and showing my approximate actual location as a red circle.
After I realised what is going on I've tested this on both my EPLs in different locations and it seems to be a problem on both. At first I thought there might be something small moving causing the issues. But after moving the sensors around and retesting I believe I've ruled that out. The sensors will consistently show targets in completely empty parts of the room. I've tested this is very repeatable once you know roughly where to stand. 
The sensors appear to work absolutely fine with everything actually within the sensor range, and I can see targets on the maps moving as expected. This is only happening just beyond the sensor range, but unfortunately this means my automations are triggering incorrectly.
[image]

Draghmar, 2024-11-03:
I wonder, does this phantom target move at all?

simonlirwin, 2024-11-03:
Yes, it does. It will hunt around a bit, just like a "normal" target does where it's re-estimating the centre of the movement. It will try to follow me if I'm moving, but it's a very small window where this issue is happening, so it will lose me pretty quickly.

EverythingSmartHome, 2024-11-03:
Does it happen when the installation angle is 0?

simonlirwin, 2024-11-03:
Yes, it does happen with installation angle of zero. This issue was also happening before that update. I hadn't reported it because I wasn't sure at that point whether it was my automations which were the issue. After I started using the dashboard map it's a lot easier to visualise what is going on.
I've just done a bit more investigating, and I think it might be the sensors themselves.
I tried updating the sensor firmware using the HKL Radar. I was hoping that would resolve the issue now they're both on the latest firmware. It might have improved things a bit, but not eliminated the problem. Updating the firmware also gave me access to the raw sensor data on the HLK app, and the problem exists showing the location wrong in the HLK app. So it's probably not related to the EPL software, but the sensor itself reporting the location incorrectly.
Has anyone managed to replicate this issue at all?
I'm wondering if there's a batch of sensors with an issue. I bought both my EPLs at the same time, so having sensors from the same batch is quite possible.

Draghmar, 2024-11-03:
At first your issue looked similar to what [I have](https://github.com/EverythingSmartHome/everything-presence-lite/issues/126), especially when you've said about this getting happen when you're leaving area, because that's the only thing I suspected in my case. But then again you've said the phantom moves and I don't think my does.

codemonkey2k5, 2025-01-31:
This is the same issue I am having.

EverythingSmartHome, 2025-02-04:
Please use the new "Stale Target Reset" feature added in 1.2.1 to address this issue. Described here and it's for this situation: https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/home-assistant-entities.html#stale-target-reset

closed 2025-02-04 by EverythingSmartHome