EverythingSmartHome/everything-presence-lite #229: Feature request: zone configuration walk

mor5cho, 2024-11-03:
As a user I'd like to create zones by selecting 1-4 (and exclusion zones), hit record and walk them up and down. Configurator is checking for my min/max x and y and creates zones on those values.

mor5cho, 2024-11-03:
Also it would be nice to create zones with 3 to 6, better 12, points, not only rectangles.

simonlirwin, 2024-11-03:
I really like the first idea. It may prove easier if there's a "set" button against each min/max which would take the current target 1 location into the appropriate field. That could make setting a zone a very fast job, without the need for creating a sort of record function. Does that sound good?

I like the second idea as well, but it sounds extremely complicated to me. The code would needing to calculate diagonal lines and take into account user input possibly creating criss-crossing co-ordinates. Personally, using 3 zones (remember you can overlap them), utilising the installation angle, plus an exclusion zone, already provides a huge amount of flexibility. If that doesn't cover your needs, possibly more rectangular zones might provide enough flexibility for you?

mor5cho, 2024-11-03:
So your suggestion is, that a user sets each corner of a zone by hitting a "set" button, right? So the user has at least four buttons hits. From a user perspective it might be easier to just hit one button to start a "record" mode and one to stop it. As a user I also see the room I'm in and the zones (f.e. couch, dining table, kitchen counter) in there. So I'd prefer to think about coordinates as less as possible and instead prefer to record the exact movements I want to keep track off.
 
Anyway, hitting "set" while being tracked at the corners of a zone would be an improvement as well and is for sure a good idea as well. But in comparison I'd say this is the more mathematical approach while the original idea is a bit more human based. 

Both ways would benefit from zones not restricted to rectangles imho. 4 cornered is fine for sure, but rectangles are a bit limiting because of the right angles. The installation angle only helps with that in rooms where every zone is in right angles. In my case the installation is in a corner and I have two zones pretty close to each other and both not in right angles. Interference is not acceptable here, so I have dead spots instead. The other two zones are in right angles. 

I'm not sure if my initial explanation was clear enough with the min/max part. I did not meant the user to set or think about those. In a "record" mode your code could track a targets x and y readings, take the minimum and maximum being read from both axles and set a zone based on those. So a user hits "record", just walks around their table for example, maybe twice to avoid misreadings, then stops the "record" and that's all.

Could be improved later by drawing more complex forms, for example L-shaped or round. When the user is not responsible for the coordinates having the correct order because your code is (you mentioned criss-crossing), it might work out. I'd also assume that drawing shapes by some coordinates is not something to be built from scratch. I'm not a programmer, though. I'm a UX/UI designer. What comes to my mind as an alternative to drawing shapes is a grid of sectors. Let's say 25 x 25 cm, named a1-z25 or just numbered for optional smaller ones/bigger rooms. When walking up in "record" mode, code records every sector a user has been tracked in and that makes a zone. An option to include the ones being surrounded by the record (for example the table itself a user was walking around) could be helpful. Also a graphical view, pretty similar to the one you have now, just with smaller grid would be great as well for users to mark or unmark every sector by hand anytime. This way any form would be possible, even zones with several parts not being connected. For example a zone including the space in front of/behind all windows/doors aka. where targets enter or leave the room. Or every chair in a room. That'll be disruptive in my (designer's) eyes!

clintebbesen, 2024-11-10:
> I really like the first idea. It may prove easier if there's a "set" button against each min/max which would take the current target 1 location into the appropriate field. That could make setting a [...]

This is very interesting to know zones can overlap, I was previously told they cant and need to have a 10mm gap between zones to avoid potential issues. I do have a use case where overlapping zones would be good, as without it I would need 6 zones to achieve what I want.

which zone takes preference when they are overlapping?

codemonkey2k5, 2025-01-31:
This is very similar to how you set up an Aqara FP2 is set up.