EverythingSmartHome/everything-presence-lite #232: Zone X Occupancy Clear when Target Count > 0 & Occupancy Detected

clintebbesen, 2024-11-10:
As seen in the screenshot below, on a daily basis I am having issues where the zone I am standing in says Occupancy = 'Clear', Target Count = 1, and the Overall occupancy sensor says 'Detected'. 

A few months ago I added 10mm gap between all zones, but it made no difference, still regularly having issues

I am running both the latest ESPHome (2024.10.3) and EST Lite (v1.2.0) firmware, I installed the non bluetooth proxy version.

[image]

Config:
[image]

clintebbesen, 2024-11-14:
This has been regularly occurring, rebooting the esp fixes it for a little while.

clintebbesen, 2024-11-19:
Similar issue I noticed today, except this time all of the occupancies were clear, target counts were all 0, but target 1 was actively tracking me while in zone 1

[image]

[image]

fidesachates, 2024-12-11:
I just received my device and have noticed this as well.

pkconnolly, 2024-12-16:
I'm having this same issue. Firmware: 1.2.0 (ESPHome 2024.11.0)
[image]

montreyw, 2025-02-01:
Having the same issue. Reboot fixes it but wondering what the other options are.