EverythingSmartHome/everything-presence-lite #237: False Occupancy detection

SteveB766, 2024-11-12:
I am getting a false Occupancy detected, I have swapped the mmWave board from another unit and the error is following the mmWave PCB.

It's claiming to detect a target 160mm away with a resolution of 258 mm and a speed of -7.72ms. These are the same reported values when I move the mmWave PCB between different sensors in different rooms.

I am assuming the mmWave board has developed a fault, this unit was purchased in October 2023 and used 24/7 since then.

[image]

EverythingSmartHome, 2024-11-12:
Hello, please update the firmware on the mmwave to 2.04 using the HLKRadarTool app, this is a known issue with the older firmware on that board. When you open the app it will scan for devices over bluetooth, if you don't see any then in Home Assistant, toggle the "mmwave bluetooth" switch to on.

SteveB766, 2024-11-12:
Thanks, that's fixed after, once I got the right app connected IOS

closed 2024-11-12 by SteveB766