EverythingSmartHome/everything-presence-lite #239: Occupancy not detected overall or on zones when it should be, but target is there and values constantly updating correctly.

clintebbesen, 2024-11-13:
Tried to setup a sensor in the bedroom, it’s not always detecting presence, currently it’s showing no presence overall or on any of the zones, however it does show target 1 as active with the correct values updating constantly. The target count of each zone is however 0. 

I was lying in bed and while watching tv the lights kept turning off, I would have to wave my arm to get it to detect me and then on the lights again. Turning my head to look at the sensor isn’t enough to trigger it, I have to wave my arm. After this happening about 3 or 4 times it didn’t work to wave my arm, checked HA and found the issue described above. 

[image]
[image]

EverythingSmartHome, 2024-11-13:
Hello! There isn't enough info in this particular case as it doesn't show zone 1 config or the exclusion zone config (if any) - could you add those?

How frequently was the target still updating?

clintebbesen, 2024-11-14:
Sorry I was in bed about to go to sleep last night so didnt want to touch my laptop. There isnt any exclusion zones for this one, and the target 1 values were right within both zones 1 and 2 but neither was showing detected. as I moved around the Target 1 values would follow me and worked correctly, it just wasn't showing detected, both zone 1 and 2 were clear.

[image]

clintebbesen, 2024-11-30:
Here is another example of the same thing where the target is moving and active, but presence isn’t shown as detected. 

https://github.com/user-attachments/assets/b512bfcf-52f5-40e0-acca-585eb01af0df

luke2844, 2024-12-11:
[image]

I've just had a similar problem. There was a person in the zone and I could see them moving in the zone configurator add on.  Home assistant reported a target count in the zone (1 in the screenshot) but occupancy was clear. The overall occupancy did show as detected though.

A restart of the EPL fixed it.

Linux685, 2025-01-20:
Same problem here.
I setup up my first EPL and yesterday was doing great. I setup my automation about 2 zones, worked like a charme. I added a third zone and it didn't work. I went to bed last night, I woke up this morning and HA show the target position, but any occupancy is detected, neither are the target count.
I saw that the mask zone had the same value of the screenshot in this post - 6000, - 1560, - 6000, - 1560, and it makes sense that is wrong, I fixed them but nothing changed.
I rebooted HA, I rebooted the sensor several times, I also tried with the factory reset, but this last i don't think worked, because I still find all the setting I put in. I'm lost...

ewanfletcher, 2025-01-25:
Just to throw my tuppence in - I've the same happening tonight .... One target detected in zone 1 (The whole room) it's not in the exclusion zone ... But no occupancy .. I can see the dot actively moving in the configurator

EverythingSmartHome, 2025-01-26:
I think we have found a solution to this but are awaiting more feedback from some users. If you'd like to try the fix, ping me a message in Discord and we can take a look. Initial reports from a few users have been very encouraging and resolved this issue, but more time is needed to say for definite