EverythingSmartHome/everything-presence-lite #241: Can't connect to ESP. Please make sure your YAML file contains an 'api:' line.

skipgub, 2024-11-16:
On my new Everything Presence Lite I have gone through the laborious process of installing ESPHome and flashing the device, but getting it into Home Assistant just won't work. It's in ESPHome but 'offline'. When I try to do anything about it I am asked to "Please enter connection settings of your [ESPHome] node." This apparently means the device? I took its IP address from my Orbi network and enter that. Port number 6053 is pre-filled. (I have no idea why or what that means but left it alone).  I get an error message "Can't connect to ESP. Please make sure your YAML file contains an 'api:' line." I tracked down a YAML file for the device, which indeed contains
# Enable Home Assistant API
api:
  encryption:
    key: "BLAHBLAH"
So, now what? There seems to be no way to get this installed and the Installation instructions peter out after assuming all will work. What if it doesn't?

EverythingSmartHome, 2024-11-17:
Can you provide logs over USB? Your DNS does not appear to work correctly on your network and so it's not connecting over wireless to get logs. USB logs would be great if you can.

Not sure what you mean with "laborious process of installing ESPHome"? Perhaps this is where the issue is, as there is no need to install ESPHome (docs do not mention it) - can you talk me through the steps you did?

skipgub, 2024-11-17:
Here's a screenshot of the instructions talking about ESPHome. 
[image]

Here's the process I followed, closely reading the instructions. I tried more than once without success. 

Installing Everything Presence Lite

From the instructions at

https://everythingsmarthome.github.io/everything-presence-lite/updating.html

My first problem was that there was no automatic setup because, I discover, my Home Assistant Green does not have Bluetooth. 

So, skipping ahead through the confusing manual setup I eventually manage to flash the device. (I'm not actually sure what this means, does, or why it's necessary because I thought I read that everything it needed was factory installed). 

Next, Connect to WiFi: no problem. [Aside: At this point the EPL is still attached by a USB cord to my Macbook. Not the cord it came with because that ends in USB-A not USB-C, but it seems to be working. I didn't see anywhere in the instructions an explanation of when you can disconnect it and put it somewhere to actually use it, nor indeed what power supply to use.]

Next is, "Device connected to the network! Add to Home Assistant". That is why we are here, yes, so I click it. 

Next it says "Open page in your Home Assistant?
You've been linked to the page that will start setting up a new integration.
Open link". I click it. 

Home Assistant now opens, and a popup says "Do you want to set up ESPHome?"

Well no, not really, but there's no other option except Cancel, whereupon the process stops. So I click OK.

Now we get the notorious "Please enter connection settings of your ESPHome node."

It wants "Host*" and has pre-filled "Port 6053". I, and numerous others from what I have read, don't understand this. What's a node? Where did the port number come from? Etc. 

Not explained in the setup instructions. What the instructions are saying meanwhile is this:

"For most people, Home Assistant will automatically discover and add the EPL immediately, thanks to ESPHome’s mDNS discovery.
To add it to Home Assistant, simply head to Settings > Devices and Services and you should see the Everything Presence Lite has been automatically discovered ...".

Well, I'm not most people. So I am into Manual Method, using Fing to try to figure out an IP address. 

Instructions: "Once you have the IP address, head over to Home Assistant and go to Settings > Devices and Services and click the add Integration button in the bottom right. ... Search for ESPHome to add an ESPHome device, and then enter the IP address into the box: ...".

And we're back to the Host* and Port dialog. 

I enter the IP I have looked up, leave Port 6053 as it is - and that just times out with the message at the start of this thread: "Can't connect to ESP. Please make sure your YAML file contains an 'api:' line."

?

skipgub, 2024-11-17:
I've been looking at the network. EPL is on IP address 192.168.1.xx on my Orbi WiFi mesh satellite while connected by USB to a MacBook Air that is on 192.168.1.xy on the same satellite; the physical Home Assistant Green is on 192.160.0.xxx/24 via a 5-port switch on the Virgin Media router (which also serves the Taco hub and Hue Bridge). My web interface for it is on the Air and works very solidly, or using the iPhone app, also excellent. 

Is that significant? I'm wondering what difference it would make if I were to plug the HA-G into the ethernet port on the WiFi hub...? For the avoidance of doubt I know zero about networking ;)

EverythingSmartHome, 2024-11-17:
It could be, depends on your network setup. You would need both your Green and Lite to be on the same network and/or accessible if they are on different subnets. Is the Lite on 192.168.1.x reachable by the Green on 192.168.0.x?

skipgub, 2024-11-17:
In what sense, reachable? It doesn't install: but what does the weird error message "Please make sure your YAML file contains an 'api:' line." mean? (Incidentally I can no longer find an ESPHome YAML file. I think I deleted it in one of my multiple attempts to get this to work).

EverythingSmartHome, 2024-11-17:
Network traffic needs to be able to pass from one network to the other. If your Green on 192.168.0.x cannot communicate with 192.168.1.x then of course it won't work. You would need to verify if this is possible.

The error is unfortunately just a red herring, it will show that message if it is unable to communicate with the IP you enter into the box.

The instructions do not mention the ESPHome add-on (where you mention you are using to create a YAML), no reason to use it if your not familiar with it

skipgub, 2024-11-17:
I can find it using Ping (ICMP) from HA on my laptop.

skipgub, 2024-11-17:
So, hang on, which ESPHome are the instructions talking about?

skipgub, 2024-11-17:
[image]

skipgub, 2024-11-17:
[image]

EverythingSmartHome, 2024-11-17:
Thanks - this is a network configuration issue for sure. You would need to ensure that the Lite is at least pingable from the Green. You may want to revisit the reason for the network being setup the way it is (not sure if you set it up or there was a particular reason)

skipgub, 2024-11-17:
I'm really sorry to be so stupid but "HAs wording for he ESPhome integration (not add-on)" makes no sense. How many ESPHomes are there?

skipgub, 2024-11-17:
I set up the network or at least, years ago we got a WiFi mesh to improve the signal in a tall narrow building. The Orbi hub plugs into the Virgin router and many, many devices connect to the WiFi. It just works and has done for years so I ave never had reason to question it ... until this one device.

EverythingSmartHome, 2024-11-17:
> I'm really sorry to be so stupid but "HAs wording for he ESPhome integration (not add-on)" makes no sense. How many ESPHomes are there?

No worries can be confusing. Home Assistant has "integrations" and "add-ons". There is an ESPHome integration (which is used to integrate the Lite into Home Assistant) and there is an ESPHome add-on (which is used to create custom configurations and firmwares for ESP32 devices)

If you think about it in its most basic form, your Home Assistant server, has to be able to communicate with the Lite (and indeed any other device you want to add) over the network. If a ping (the most basic form of network communication) is failing then generally it means one of two things:

1. The IP address your trying to ping isn't correct
2. The Network doesn't permit access, either through misconfiguration or it's intentionally blocked with firewall rules or network access rules

skipgub, 2024-11-17:
That makes sense but why - out of dozens of devices - is it just this one? Looking at the list of Attached Devices, how is HA seeing (for example) Ring cameras that are on the WiFi mesh, but not this EPL?

EverythingSmartHome, 2024-11-17:
Could be the first one? Are you sure the IP you are providing is correct?

skipgub, 2024-11-17:
It's the IP shown in the list of attached devices. But ... wait ... Ah! 
[image]
But ... 
[image]

EverythingSmartHome, 2024-11-17:
Also it seems that the ring integration is cloud based, so it wouldn't need to communicate directly with them as they go through the cloud (at least from what I can see, I don't use them)

skipgub, 2024-11-17:
All that changed for the EPL to be Discovered was that I removed it from the laptop's USB port and powered it up independently on a USB socket (is that appropriate, by the way?).

skipgub, 2024-11-17:
What is this Port 6053 thing? Is it correct?

skipgub, 2024-11-17:
What does "Discovered" mean if I then have to enter its IP address (and it doesn't find it)? This is all bewildering ...

skipgub, 2024-11-17:
OK the documentation for the Discovered device leads me to this: https://esphome.io/components/api.html.  I can't understand any of it, but can someone tell me what "api:" I need to put where - which YAML?

EverythingSmartHome, 2024-11-17:
You don't need to add any API line as I said before.

The port is the network port used for communication, it can be ignored.

You don't need to enter an IP if it's discovered, the IP is only if your network is not setup in a way to allow discovery

skipgub, 2024-11-17:
But then what is the "Add" button for on the Discovered device? How else do I do anything with it?

EverythingSmartHome, 2024-11-17:
If it's discovered then you just press the add button as per the screenshots in the docs. The IP address is only if it's not discovered and you need to add it manually

skipgub, 2024-11-17:
When I press Add, this is the popup I get. 
[image]
... which thinks about it for half a minute, then produces the line about needing an api which is where this whole thing began :)  I have to say, the ESPHome documentation linked off the discovered device talks about using a native ESPHome API, and in turn links to a very long page that I can't understand. I know you say it isn't required but, well, it does keep asking me for it. https://esphome.io/components/api.html

EverythingSmartHome, 2024-11-17:
Indeed but the API line is a red herring and in 99% of cases it's produced when Home Assistant can't communicate with the device your trying to add.

If Home Assistant can't ping the Lite, it will never be able to add it

skipgub, 2024-11-17:
Then how did it ever Discover it? I guess it's just another aspect of Home Assistant that makes no sense: I am coming to find these are legion! Thanks very much for trying to help anyway, I really appreciate your time.

skipgub, 2024-11-17:
Hurray! What did I do? Ab so lute ly nothing. In fact I was cooking dinner. But suddenly it was there. Go figure, as they say.

[image]

skipgub, 2024-11-17:
Oh well. Fun while it lasted. I'll try again tomorrow. 
[image]

skipgub, 2024-11-18:
Adding a final entry to explain how I fixed this, if anyone is interested. I discovered that *both* my Virgin router and my Orbi mesh are broadcasting WiFi using the same SSID and password. Seamless to users, but it meant devices could be on one of two networks. In this case, HA on the Virgin and the EPL on the mesh, so not being seen by it. I temporarily turned off the mesh, whereupon EPL connected on a Virgin network IP. I then made that static [read: learnt how to reserve an IP], and fingers crossed everything seems to be working.

EverythingSmartHome, 2024-11-18:
Glad to hear you got the network figured out, thanks for the update! I'll close this off for now as sounds like you've resolved.