EverythingSmartHome/everything-presence-lite #242: EP Lite Unavailable in Home Assistant after short issue with internet connection

Valve90210, 2024-11-16:
Yesterday evening, my router dropped my internet connection for a minute or two until I rebooted it and it reconnected.  Later in the evening, I went into my kitchen an my lights, which are controlled by my EP Lite, didn't come on. I looked at the EP Lite and noticed a solid red LED. 

Checking in Home Assistant, it was showing as unavailable and had been since around the time the internet dropped. 

I powercycled the EP Lite. The LED flashed a bit then stopped. However it continued to show as unavailable in Home Assistant. I tried Rebooting HA and that didn't help, then I rebooted both again and still it's showing as unavailable. 

Looking at my router, the EP Lite is connected to my wifi on the correct IP address, and using Fing I can ping it. 

However, looking in my HA logs I have the following error warning:

Logger: aioesphomeapi.connection
Source: runner.py:189
First occurred: 00:02:32 (715 occurrences)
Last logged: 20:39:28

everything-presence-lite-aa2d08 @ 192.168.1.117: Connection error occurred: everything-presence-lite-aa2d08 @ 192.168.1.117: Connection lost

I'm at a bit of a loss as to what has happened, and how I can get it working again?

EverythingSmartHome, 2024-11-17:
Hello, can you provide logs over USB instead of wireless? Unfortunately they don't give details since it's a connection problem

Valve90210, 2024-11-17:
> Hello, can you provide logs over USB instead of wireless? Unfortunately they don't give details since it's a connection problem

Might you be able to talk me through what exactly I need to do for this? Plus I to usb on home assistant or on my pc?.

And then what do I need to do?