EverythingSmartHome/everything-presence-lite #245: Everything Presence Zone Configurator not correctly updating position.

superiormirage, 2024-11-20:
I have two Everything Presence Lite that are exhibiting the same behavior. When I try and use the Zone configurator add-on for home assistant, my position isn't updating correctly on the radar map. 

No matter where I move, the red dot always stays in the center of the radar, right in front of the sensor. 

[image]

I believe the Lite is updating my position correctly, but it's not showing up on the visual radar.

I was in the same spot as the screenshot above.

[image]

EverythingSmartHome, 2024-11-21:
This is because the units of measurement have been changed from the default of mm to inches. The zone configurator expects mm (default for the Lite)

packetwarrior, 2024-11-29:
Maybe I'm a little dense here, but I keep seeing this answer everywhere, but no details about how to change it back to metric.  We aren't talking about changing the 'Unit system' in Home Assistant's system setting are we?  Can't find any similar setting anywhere else.  I just bought a few more sensors, so I'm hoping that's not the case.

billyjoebob999, 2024-11-29:
> Maybe I'm a little dense here, but I keep seeing this answer everywhere, but no details about how to change it back to metric. We aren't talking about changing the 'Unit system' in Home Assistant's [...]

Having the same problem. To change the unit of measurement I just clicked the the sensor. Click the gear icon and change the unit of measurement from `in` to `mm`

[image]

Did this for each sensor

[image]

Jord4n, 2024-12-03:
Thank you for the fix solution! Annoying we need to change all of these though, ha.

EverythingSmartHome, 2024-12-03:
Some users tell me there is too many sensors and most should be hidden, and some users tell me they want all of them to be visible with none hidden - I can never please everyone unfortunately 😅

x0SiN0x, 2025-02-16:
Did you have to do something in order for the change in measurement to work?  I've set all mine to mm and still no go, even restarted but same thing (the red dot is always at the top of the box)

EverythingSmartHome, 2025-02-16:
The configurator supports both mm and inches now so no need to change it. Make sure to update your Lite, should be on 1.2.2

x0SiN0x, 2025-02-16:
Thanks for the update, i show i'm on version 1.2.2 but still not working.  Will reset all the settings ive been playing with to try and get this to work and see if it fixes itself.

x0SiN0x, 2025-02-16:
Reset things and still not working.  But tried setting up a plot card that i found online and shows the motion is behind, what setting would adjust this?  All settings should be reset back to defaults so not sure what that could be.

[image]