EverythingSmartHome/everything-presence-lite #246: WARNING Can't connect to ESPHome API for everything-presence-lite

BrokenEyesCode, 2024-11-21:
I'm not sure this is the right place to ask for support, but if not, please point me in the right direction.

My EPL became unavailable one evening, not 100% sure why, might have been an update, but ever since it will not connect.
Assuming it was an update / firmware issue, I've been trying to fix it.
I have created a new config in ESP Home and tried to install on the EPL.
While it 'compiles' and I get 
```
======================== [SUCCESS] Took 424.79 seconds ========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.142
INFO Uploading /data/build/everything-presence-lite-xxxx/.pioenvs/everything-presence-lite-5c060c/firmware.bin (1557824 bytes)
Uploading: [============================================================] 100% Done...
```
I then get this in the logs
```
INFO Successfully connected to everything-presence-lite-5c060c @ 192.168.0.142 in 11.230s
WARNING Can't connect to ESPHome API for everything-presence-lite-5c060c @ 192.168.0.142: Handshake timed out after 30.0s (TimeoutAPIError)
INFO Trying to connect to everything-presence-lite-5c060c @ 192.168.0.142 in the background
WARNING everything-presence-lite-5c060c @ 192.168.0.142: Connection error occurred: everything-presence-lite-5c060c @ 192.168.0.142: Connection lost
INFO Successfully connected to everything-presence-lite-5c060c @ 192.168.0.142 in 0.054s
WARNING everything-presence-lite-5c060c @ 192.168.0.142: Connection error occurred: everything-presence-lite-5c060c @ 192.168.0.142: Connection lost
INFO Successfully connected to everything-presence-lite-5c060c @ 192.168.0.142 in 0.015s
WARNING everything-presence-lite-xxx @ 192.168.0.xxx: Connection error occurred: everything-presence-lite-5c060c @ 192.168.0.142: Connection lost
```'
I can ping it, but that's it.
The Red Led flashes 7 times if that is meant to indicate anything.
Thinking this was a flashing issue, I tried to connect the EPL to my computer (Ubuntu Linux 24.04) using Chrome, but that failed, saying it couldn't open the usb port.

So what can I do next to fix this?
Thanks
Leona.

BrokenEyesCode, 2024-11-21:
Well it came online for a minute, then went back to unavailable :(

EverythingSmartHome, 2024-11-23:
Are you able to upload the logs over USB?

BrokenEyesCode, 2024-11-24:
> Are you able to upload the logs over USB?

Yes I have access to the logs, attached.
[epl_logs.txt](https://github.com/user-attachments/files/17894475/epl_logs.txt)

EverythingSmartHome, 2024-11-24:
It looks like your running into a bug in ESPHome that causes a boot loop when internet is blocked (or not accessible) and it tries to check for GitHub for updates - is this the case? A common one is primary DNS server not functioning properly and ESPHome doesn't appear to use the secondary

EverythingSmartHome, 2024-11-24:
A re-flash won't change anything if issues are still present unfortunately

BrokenEyesCode, 2024-11-24:
> A re-flash won't change anything if issues are still present unfortunately

Ok I believe I have resolved the DNS settings issue, is there a way to stop it boot looping?

EverythingSmartHome, 2024-11-24:
When the issue is resolved it will stop next time it loops. Do keep in mind that ESPHome only appears to use the primary DNS server and doesn't use the secondary DNS server just incase that was part of your fix

BrokenEyesCode, 2024-11-25:
> When the issue is resolved it will stop next time it loops. Do keep in mind that ESPHome only appears to use the primary DNS server and doesn't use the secondary DNS server just incase that was part of your fix

Can I tell which DNS server the ESPHome is using so that I know if my settings have taken effect?  Is there a way to check/change the config?

EverythingSmartHome, 2024-11-25:
According to the last logs you uploaded it says:

```
[C][wifi:436]:   IP Address: 192.168.0.142
[C][wifi:441]:   Hostname: 'everything-presence-lite-5c060c'
[C][wifi:448]:   Subnet: 255.255.255.0
[C][wifi:449]:   Gateway: 192.168.0.1
[C][wifi:450]:   DNS1: 192.168.0.201
[C][wifi:451]:   DNS2: 192.168.0.1
```

EverythingSmartHome, 2024-11-25:
It's from your DHCP server I'm guessing (it's a network thing not a Lite thing so hard to say without seeing your network)

BrokenEyesCode, 2024-11-26:
Ok, I think I might have fixed it, bit of a workaround, but on the last re-flash, I chose not to configure the Wifi and let HA find it over BT, unplugged it from the laptop and plugged it into the power in the location it should be in.  Let HA find it, and add the Wifi details that way, then HA was happy it could see it, and so far, it's working. 
So the fix is, refrash, don't configure the wifi, allow HA to find over BT and then configure Wifi, what a pain, but we got there, thank you.
I would say, there is something amiss with the wifi module or how to stores the config, because I saw 'authentication' errors in the logs, when it tried to connect to my wifi and my details where certainly correct.  As it's happy with my DNS settings now without me changing anything.

closed 2024-11-26 by EverythingSmartHome

deancommasteven, 2024-12-21:
@EverythingSmartHome Is this something that will be fixed? I block all IoT devices from internet access and do not want each device to be able to connect directly out. This bug means I cannot use Everything Presence sensors unless I give them full-time access directly to the internet, which I am not willing to do. Should I file a report with ESPHome? These sensors are amazing otherwise, thank you for making them.

TripelJ, 2024-12-22:
> @EverythingSmartHome Is this something that will be fixed? I block all IoT devices from internet access and do not want each device to be able to connect directly out. This bug means I cannot use [...]

While you wait for the official reply :) 
One suggestion. If you don’t want to have your IoT to the internet. You can always have the YAML setup manually. Right now it’s pointing to GitHub for the easy of setup and updates. 
But if you don’t want you can just put the config file together with all the pieces you need and remove GitHub. And then each time there is an update you will have to merge it with your config by hand. 
But that is one way to do it. 
 
Cheers.

deancommasteven, 2024-12-22:
> While you wait for the official reply :) One suggestion. If you don’t want to have your IoT to the internet. You can always have the YAML setup manually. Right now it’s pointing to GitHub for the [...]

Ah, I did forget I could do that. Thanks for the suggestion! I would ultimately like to use the default firmware, but that's a good stopgap.

EverythingSmartHome, 2024-12-22:
I hope so but it's for ESPHome to fix unfortunately, not something I can do on my side.

One thing you can try is to change the way you are blocking. Are you blocking it via the "absence" of a firewall rule?

deancommasteven, 2024-12-22:
> I hope so but it's for ESPHome to fix unfortunately, not something I can do on my side.
> 
> One thing you can try is to change the way you are blocking. Are you blocking it via the "absence" of a firewall rule?

I probably should be allowing instead of blocking, but no, I have blocking rules for specific devices. How should I block the default firmware it so that it doesn't boot loop? I have control over everything including DNS.

EverythingSmartHome, 2024-12-22:
What type of action do you take, drop it reject?