EverythingSmartHome/everything-presence-lite #247: It doesn't work at all

citarone, 2024-11-27:
It doesn't work at all—no recognition when I'm standing in front of it, moving, or doing anything else. The detection range is set to 100 cm, but it occasionally detects ghosts, while it doesn't recognize me at all. I want to set it up, configure it, and then forget about it—not spend forever tinkering with it just to maybe get it working. Now I've spent 50 euros for nothing.

citarone, 2024-11-27:
do you mean this 
[image]

EverythingSmartHome, 2024-11-27:
Yes please, plus all the settings below that too.

Can you advise order number?

citarone, 2024-11-27:
my oder number is #EST22125  

I didn’t change anything, and now it seems to be working. I have no idea why it suddenly works. so sorry for my rant 
[image]

closed 2024-11-27 by citarone